realsenseai/librealsense #3663: S/N detection discrepancy: rs-enumerate-devices and udev

130s, 2019-04-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 5.11.1.100 | 
| Operating System & Version |   Linux Ubuntu 16.04 64b  | 
| Kernel Version (Linux Only)    |  4.15.0-47-generic |
| Platform                                 | PC |
| SDK Version                            |  2.19.2                           | 
| Segment			|  Robot                          | 

### Issue Description

I'm seeing different serial number returned from `rs-enumerate-devices` command and `udev` (2 Realsense devices are connected to our computer).

```
$ rs-enumerate-devices | grep Serial
    Serial Number                 :     839112062356
    Serial Number                 :     839112061727
```
```
$ lsusb|grep -i 8086
Bus 002 Device 008: ID 8086:0ad3 Intel Corp. 
Bus 002 Device 007: ID 8086:0ad3 Intel Corp.

$ dmesg|grep -i serial|grep -i 2-7
[ 6422.188134] usb 2-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6422.188145] usb 2-7: SerialNumber: 841113021535
$ dmesg|grep -i serial|grep -i 2-8
[ 6423.643856] usb 2-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6423.643859] usb 2-8: SerialNumber: 841113021627
```

Probably because of these discrepancy, I'm getting errors in the [downstream intel-ros/realsense](https://github.com/intel-ros/realsense/blob/a70edc7364252224ccbfcab810ed09a4d58b9cc9/realsense2_camera/src/realsense_node_factory.cpp#L70) when I pass the values I got from udev:
```
The requested device with serial number 841113021535 is NOT found!
```

Some more environment info:
```
$ apt-cache policy librealsense2|grep -i Installed
  Installed: 2.19.2-0~realsense0.775
$ apt-cache policy librealsense2-dkms|grep -i Installed
  Installed: 1.3.5-0ubuntu1
$ apt-cache policy librealsense2-utils|grep -i Installed
  Installed: 2.19.2-0~realsense0.775
$ modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.5
srcversion:     3286F64D5B1431E64E8C37C
```

dorodnic, 2019-04-04:
Hi @130s 
While this is true, I doubt it is going to be changed (the USB serial is serial number of the ASIC board, while the LRS serial is of the camera head).
Is it problematic to use the output of `rs-enumerate-devices -s`?

130s, 2019-04-08:
@dorodnic Thank you for clarification. I didn't know that.

>  Is it problematic to use the output of rs-enumerate-devices -s?

No it isn't now that I know the correct answer.

I'd say this setting might be inevitably confusing, and I understand this is what it is. It is worth being more clarified, maybe in a documentation. Also better error message can help. I suggest to keep this ticket open before some solution is implemented.

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

Hi 130s,

Thanks for the feedback and suggestions. We will update in the github if any.
Please check back later.

Cheers!

dorodnic, 2019-07-22:
We have added ASIC_SERIAL_NUMBER to librealsense to reflect the SN appearing in lsusb. 
We also considered aligning SDK S/N to libusb convention, but I fear that would have been to intrusive of a change for many customers.

closed 2019-07-22 by dorodnic