realsenseai/librealsense #3664: D435i on jetson TX2 with usb3

jnre, 2019-04-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | -| 
| Operating System & Version |    Linux 18.04   | 
| Kernel Version (Linux Only)    |  4.9.140-tegra | 
| Platform                                 |  NVIDIA Jetson TX2  |
| SDK Version                            |  {  2.19.2} | 
| Language                            |  C++ | 
| Segment			|  Robot                           | 

### Issue Description
im having issues trying to connect D435i with usb3, usb2 is fine.

using D435 works for both, is it due to kernel version

dorodnic, 2019-04-03:
Should be Ok (I tested this just couple of days ago, without any modifications to the kernel)
Please attach regular logs (dmesg when connecting the camera, lsusb)
We will do our best to help

jnre, 2019-04-03:
dmesg:
[16514.426406] pcieport 0000:00:01.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, id=0008(Receiver ID)
[16514.436687] pcieport 0000:00:01.0:   device [10de:10e5] error status/mask=00000001/00002000
[16514.445095] pcieport 0000:00:01.0:    [ 0] Receiver Error         (First)

continuing down,
[16514.452972] pcieport 0000:00:01.0: can't find device of ID0020
[16515.213403] usb 2-2: USB disconnect, device number 8
[16515.213424] usb 2-2: Failed to suspend device, error -19
[16515.218887] usb 2-2: usb_suspend_both: status -19
[16515.417480] usb usb2: usb_suspend_both: status 0
[16522.529579] usb 2-2: new SuperSpeed USB device number 9 using xhci_hcd
[16522.558034] usb 2-2: New USB device found, idVendor=8086, idProduct=0b3a
[16522.558045] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16522.558052] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435i
[16522.558057] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[16522.558061] usb 2-2: SerialNumber: 852323053945
[16522.563646] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[16522.563655] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[16522.564998] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[16522.572755] uvcvideo: Unable to create debugfs 2-9 directory.
[16522.572908] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[16522.582784] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[16522.590810] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[16522.600528] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[16522.608625] input: Intel(R) RealSense(TM) Depth Ca as /devices/10003000.pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0/usb2/2-2/2-2:1.0/input/input67
[16522.611123] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[16522.611134] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[16522.615262] uvcvideo: Unable to create debugfs 2-9 directory.
[16522.615435] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
[16522.623430] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[16522.631367] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
[16529.917250] usb 2-2: usb_suspend_both: status 0
[16529.917338] usb usb2: usb_suspend_both: status 0
[16535.165267] usb 2-2: usb_suspend_both: status 0
[16535.165376] usb usb2: usb_suspend_both: status 0
[16539.997253] usb 2-2: usb_suspend_both: status 0
[16539.997337] usb usb2: usb_suspend_both: status 0

lsusb:
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 008: ID 0bf8:101e Fujitsu Siemens Computers 
Bus 003 Device 007: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 003 Device 006: ID 214b:7250  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 009: ID 8086:0b3a Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

thanks for helping!

RealSenseCustomerSupport, 2019-04-17:
----------------------------------------------

Hi @jnre,

You can also refer to #3165 which is a similar issue. Make sure HID module installed ready for D435i IMU.

koenekle, 2019-05-16:
@RealSenseCustomerSupport 
Our team has a a simmilar Issue as jnre. 
Unfortunately the scripts of jetsonhacks are for the 4.4 kernel and not for 4.9. 
We tried to apply the patchfiles to our kernel, but this failed. Are there any patchfiles which will work on the 4.9 kernel?

RealSenseCustomerSupport, 2019-05-31:
----------------------------------------------

Please refer to #4004 to see if it can resolve the same issue.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @jnre, @koenekle,

Is the issue resolved. I will close this ticket if no additional question for this topic.

koenekle, 2019-11-05:
We worked it out by using this Version of the Realsense kernel patches: 
https://github.com/Tengyun-Mo/buildLibrealsense2TX2

However it would be a great support, if you could update your [docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
or at least mark them as deprecated, and link to external sources like this one: 
https://github.com/JetsonHacksNano/installLibrealsense
(Even though this is for the Jetson Nano, the Nvidia Kernel for all Jetson Boards should be the same)

closed 2019-12-05 by RealSenseCustomerSupport