realsenseai/librealsense #3670: T265 hardware synchronization

aharmat17, 2019-04-03:
Is there a way to synchronize a T265 with the D-series cameras? We'd like to trigger the D-series from the T265. I know it doesn't have an easily accessible hardware port, but maybe we can connect to something on the T265's circuit board?

MartyG-RealSense, 2019-04-03:
This is a somewhat difficult subject.  I think that your 400 Series camera would have to be set as a 'slave' camera, and whatever is generating the sync pulse that is sent to the slave acts as an external trigger mechanism instead of having another 400 Series camera as a 'master' camera.

The link below has a list of links regarding using an external trigger.

https://forums.intel.com/s/question/0D70P000005PX0iSAG

As you mentioned, the T265 does not have a set of hardware sync pins like the 400 Series cameras have.  So whatever you connected onto the T265 circuit board would either have to send some kind of simple pulse to the 400 Series camera along a wire, or transmit the pulse wirelessly.

aharmat17, 2019-04-03:
Thanks Marty, yes I understand that opening the T265 would void the warranty, we're just trying to figure out our options. Do you know if the T265 has circuit board pins that could be used as a trigger out for the 400 series cameras? Maybe it even has the pads for a hardware header like the 400 series, but they just aren't exposed outside the case?

MartyG-RealSense, 2019-04-03:
The T265 has a data sheet document of its own, though it does not have circuit board details.  I could not find a picture of a bare circuit board for the T265 and so I do not have details of its internal hardware unfortunately.

https://www.intel.com/content/www/us/en/support/articles/000032422/emerging-technologies/intel-realsense-technology.html

bfulkers-i, 2019-04-04:
@aharmat17 the way I would suggest approaching this is from the software side.

The T265 has host-device clock sync, so the timestamps of the poses are provided in the synced system clock domain and should be fairly accurate (depends on your exact USB setup, but for sake of discussion say ~1ms).

To get a pose for the exact moment of a depth frame, instead of triggering both at the same time the best way to do it is to use the velocity and acceleration provided with the pose to predict the pose at the timestamp of the depth frame. There is an example showing how to predict the pose for the current system time here which should be a good starting point:

https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-predict

closed 2019-05-23 by RealSenseCustomerSupport