realsenseai/librealsense #3672: Unable to Disable RS2_STREAM_INFRARED

vhanded, 2019-04-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 module                                         | 
| Firmware Version                   | 5.11.1.100                                               | 
| Operating System & Version |   Win 10                                                  | 
| Platform                                 | PC                                                           |
| SDK Version                            |  2.19.0                                                    | 
| Language                               |  C++                                                       | 
| Segment			         | other | 

### Issue Description
Hi, I am using D410 module (variant without color). I called

```
conf.disable_stream(RS2_STREAM_INFRARED); 
enable_stream(RS2_STREAM_DEPTH);
```
However, when I call:

`poll_for_frames(frameset);`
I got 2 frames in each frameset (depth + infrared).

Is Infrared stream always enable, when depth stream is enabled?

Thanks.

dorodnic, 2019-04-04:
Please attach the full code snippet.

vhanded, 2019-04-04:
Hi,

I had found the problem:

Even after I disabled all the streams, and only enable depth stream, but whenever I requested any data that requires color/infrared, the stream will be activated automatically.

```
rs2::pipeline pipe;
rs2::config c;
c.disable_all_streams();
c.enable_stream(RS2_STREAM_DEPTH, RS2_FORMAT_Z16, 30);
pipe.start(c);
rs2::frameset data = pipe.wait_for_frames();
cout << data.size() << endl;
```
The above code will show only 1.

```
rs2::colorizer color_map;
rs2::pipeline pipe;
rs2::config c;
c.disable_all_streams();
c.enable_stream(RS2_STREAM_DEPTH, RS2_FORMAT_Z16, 30);
pipe.start(c);
rs2::frameset data = pipe.wait_for_frames().apply_filter(color_map);
cout << data.size() << endl;
```
The above will show 2.

closed 2019-04-07 by dorodnic