realsenseai/librealsense #3675: Repeated IMU Timestamp

KitKat7, 2019-04-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    | 4.15.0-47-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.19.2                        | 
| Language                            | N/A | 
| Segment			| Robot | 

### Issue Description
I was trying to collect some data from D435i, with both images and IMU data.
But I found the timestamps of IMU data in the collected bag were incorrect (repeated). A figure with acceleration timestamps plotted is shown below. Similar problem existed for gyroscope data.
[image]
I tried the `realsense-viewer` and the [ROS wrapper](https://github.com/intel-ros/realsense) to collect the data. But both provided the same results. I guess the problem might be from librealsense. I'd appreciate any help :).

dorodnic, 2019-04-04:
Do you know if you are running on a patched kernel?

KitKat7, 2019-04-04:
Hi @dorodnic thanks for your reply. I'm not sure how to check whether I'm running on a patched kernel, but I didn't manually do kernel patching.
The results from `cat /etc/os-release` is shown as following:
```bash
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

I tried previous [`SDK with version 2.19.1`](https://github.com/IntelRealSense/librealsense/tree/da6d553ba204ba08ca8090b12adfad91e5147a52) as suggested in https://github.com/intel-ros/realsense/issues/703. This problem seems not appeared.

ev-mp, 2019-04-10:
@KitKat7, thanks for the report.
We confirm that IMU 3-second wraparound is a bug in V2.19.2 and will be fixed in the next release.

ev-mp, 2019-04-23:
The fix is available in [v2.21.0](https://github.com/IntelRealSense/librealsense/tree/v2.21.0) release

KitKat7, 2019-04-23:
@ev-mp, this bug disappeared in v2.21.0. Thanks for the fix :).

closed 2019-04-23 by KitKat7