realsenseai/librealsense #3676: problem with usb and firmware

scizors, 2019-04-04:
| | |
|---|---|
|**librealsense**|2.19.0 RELEASE|
|**OS**|Windows|

I am using intelrealsence D435 in windows 10 computer and I am facing lots of problems 
1. USB 3.0 port is showing as USB.2.0 in Intel Real-seance viewer V2.19.0 
2.Driver problem in the Intel Real-seance viewer V2.19.0 
3.If I install the driver in my windows 10 Enterprise PC with the software intel-realsense-dfu.exe .It is i5 6th Generation which is satisfied the requirement given in the official data-sheet provided by the Intel Real-seance. I also tried with old and the latest version of .bin file.

[image]
[image]
[image]

MartyG-RealSense, 2019-04-04:
1.  The camera may register as USB 2.0 when in a USB 3.0 port if it is inserted into the USB port slowly.  Inserting the camera cable into the USB port quickly and firmly will reduce the chance of this mis-identification occurring.

2.  Please describe your driver problem with the RealSense Viewer in version 2.19.0.

3.  With RealSense 400 Series cameras such as the D415 and D435, the generation of your processor does not matter, as the 400 Series can work with any Intel or ARM processor.  The generation requirement applied to the earlier RealSense SR300 camera.

The DFU updater tool requires a USB 3.1 port.  So if your port is only USB 3.0, this could also be an explanation for your problem with the updater.

irvine48, 2019-04-07:
I am try to extend my cable for my camera D415, but i realize it did't work on typical old cable direct extension.

I found the maximum length of cable can be use to provide max data throughput.
> https://community.cypress.com/docs/DOC-10693
for USB3.2 < 1m, in this case is not possible to extend the cable longer.

my question is how much data need for this camera (default setting) to maintain the camera output. 450Mbits? 1Gbits? 3A? 5A? where can i confirm the output is sufficient (without lost frame) on this cable.

if this module with ethernet  maybe will be much easy to extend it?

MartyG-RealSense, 2019-04-07:
As a general rule of thumb, unless it is a premium high-grade cable such as an industrial-grade one, 2 meters may be the best that can be achieved before the camera stops working with it, and 1 meter would be better.

It is possible to extend the camera with additional extender equipment, such as the industrial-grade (though not inexpensive) gear available from Newnex.

https://www.youtube.com/watch?v=GLQgR1jT04M

Intel's multiple camera white paper has a table on page 6 of bandwidth use estimates for camera quantities ranging from 1 to 6 on a setup where the cameras are not hardware-synced.

  https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf

scizors, 2019-04-16:
I am using hub its USB type C cable to connect in my laptop T480 and its now working fine.

Name :USB -C 3.1 with slim hub 4 port

closed 2019-04-16 by scizors

MartyG-RealSense, 2019-04-16:
Thanks so much for the update and the information about what worked for you!

irvine48, 2019-04-23:
hey guys, here attached with the cable i using. 3meter usb-a to usb-c max resolution 640x480 (USB2.1). I am happy with it so far, add on "low heat" generate on camera.

> https://www.lazada.com.my/-i453445346-s692753242.html?urlFlag=true&mp=1

or just searching

> "ugreen" brand

 from any china shoping website (alibaba, taobao, lazada) will do.

MartyG-RealSense, 2019-04-23:
Thanks so much for your recommendations!   It is very useful to have additional options for RealSense users to explore.