realsenseai/librealsense #3678: low dept measure accuracy

maxrodkin, 2019-04-04:
[image]
| | |
|---|---|
|**librealsense**|2.19.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|746512070331|
|**Firmware Version**|05.10.03.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|2.1|

Please provide a description of the problem

hello. cam doesnt resolve drugs in palette. the distanse to palette is 30 cm.
what to set?

dorodnic, 2019-04-04:
Hi @maxrodkin 
The camera is connected via USB2, is this intentional?
Also, the selected preset is High Density, did you try with High or Medium Accuracy?
Last, what is the situation in the 3D view? The 2D colors can be misleading

maxrodkin, 2019-04-05:
hi @dorodnic 
USB2 was my fault, now it is USB3
had set preset to High Accuracy. Got a little bit better view of blister. But obviously, the accurace is not enough to recognize a drill or another working instrument.. My goal is getting the cloud of points of object and comparing to 3D model, finding diffs.
[image]
[image]

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

@maxrodkin Noticed that you adjusted the exposure and laser power also, so did you try the default setting with default preset?

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@maxrodkin Didn't get response from you for several weeks. Will close the ticket at this point. Please feel free to open new tickets if you still have questions. Thanks!

closed 2019-05-20 by RealSenseCustomerSupport