realsenseai/librealsense #3679: Setup photos of

vhanded, 2019-04-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410                                                       | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.0                           | 
| Language                            | C++                          | 
| Segment			|  other                          | 

### Issue Description
Hi, I want to synchronize all my 4 devices using HW sync, I had read the documentation, but I am not from electronic background, so can't really understand the electronic schematic drawing.

I purchased the required connector and pin. Do I need PCB board or soldering tool?

Anyone has photos of their setup?

Thanks.

vhanded, 2019-04-09:
Anyone? It seems like people do successfully setup, mind to share some photos?

Thanks.

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi

Here is a typical Sync Cable that we built for internal use. This was cut and modified from a USB2.0 cable.

Hope it is helpful.

Attachment(s):
Sync_cable.jpg - https://realsensesupport.intel.com/attachments/token/57aUyVleYuVhuLFCbOWVpPbf2/?name=Sync_cable.jpg

vhanded, 2019-04-29:
Hi @RealSenseCustomerSupport , I am not authorize to view the image from realsensesupport website. Is it possible to post it here?

Thanks.

niedan, 2019-04-29:
Hope this one helpful ......
[image]

closed 2019-05-11 by RealSenseCustomerSupport