realsenseai/librealsense #3680: Error while recording. Bag files are missing

holowgen, 2019-04-04:
| | |
|---|---|
|**librealsense**|2.18.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|801312071062|
|**Firmware Version**|05.11.01.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

I tried to record using the intel realsense viewer. When i pressed on Record, it displays this error:

UNKNOWN in rs2_create_record_device_ex(device:00000196599E0EB0, file:C):
Error opening file: C:\Users\username/20190404_164457.bag

I dont have this bag file under the directory. What should I do?

dorodnic, 2019-04-04:
Hi @holowgen 
There is probably a permission issue writing to that folder. 
In the Settings menu you can modify the default location where files are being saved.

holowgen, 2019-04-05:
the problem is I cant locate the bag file. I searched throughout and this file is missing. How can i find or create one?

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi holowgen,

To your error message (Error opening file: C:\Users\username/20190404_164457.bag), it seems to be that someone set it wrong before you started recording a .bag file (if it was not changed by you).

By default, it would be set to following path in Windows style slash (not using Linux style backslash):
C:\Users\<My_Windows_user_name>\Documents\

As for adjusting .bag file saving path, please click on the Wheel Shape icon off top-right corner in Realsense Viewer Windows. It will bring up a settings menu with 3 tabs (Playback & Record, Performance, and General). You will find the .bag saving path configuration under "Playback & Record", and change to whichever you prefer.

Hope this can help clarify. Please let us know if you still have issue.

Thanks!

holowgen, 2019-04-15:
Thank you. Your suggestion worked.

closed 2019-04-15 by holowgen