realsenseai/librealsense #3689: Better way to convert .bag file to images? different frame numbers obtained for rgb and depth when using rs-convert.  Loss in frames.

ankith001, 2019-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2019-04-05:
The link below is an excellent discussion in February 2019 of this subject:

https://github.com/IntelRealSense/librealsense/issues/3208

ankith001, 2019-04-05:
@MartyG-RealSense 
The rs_bag2image tool mentioned in the #3208 will that help in synchronizing the RGB and depth frames? I also got to know that it converts the depth data to gray-scale images. Is there another approach to get it back to the standard colored depth data?

MartyG-RealSense, 2019-04-05:
@UnaNancyOwen would be the best person to ask about using color in rs,_bag2image.  Hopefully now I have linked Una into this discussion, they can provide advice.

In regard to depth and color sync, I would recommend looking at the wait_for_frames instruction, which is discussed in regard to sync in the documentation link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#api-changes

ankith001, 2019-04-06:
How can I sync the depth and RGB frames after using rs-convert? Can you provide the following steps to do that? @MartyG-RealSense

MartyG-RealSense, 2019-04-06:
I do not have enough knowledge of this subject to be able to offer further advice, unfortunately.  Hopefully another person on this forum can help.  I do apologize.

ankith001, 2019-04-06:
@dorodnic Can you help me with the issue of synchronizing the RGB and Depth frames? How can I use rs2::syncer after rs-convert? Or any other approach you would suggest me to synchronize?

ankith001, 2019-04-11:
How to synchronize the depth and color images? I know there will be a drop in frames for sure. However, I want to see if I can synchronize the frames.

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Hi @ankith001,
How about to use the timestamps and frame counters provided in the frame metadata to help to synchronize the color and depth?
In addition, you can slightly modify the record-playback sample to record synchronize depth/color frames.
https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback
Thank you.

ankith001, 2019-04-24:
I have long videos to synchronize. The videos are around 30-40 minutes long.

ankith001, 2019-04-25:
@dorodnic @RealSense-Customer-Engineering  @MartyG-RealSense @UnaNancyOwen 
What code do I need to use to get timestamps, frame number and convert bag files to png all at the same time?

ankith001, 2019-04-30:
Hi @YangJiao1996 Did you get your code working for the rs-convert to synchronize the RGB and depth frames? Can you share it with us?

YangJiao1996, 2019-04-30:
Hi @ankith001,
Unfortunately, you cannot synchronize the RGB, depth and/or IR frames after the .bag file is generated. Currently (after the 2.20 version) what both `rs-convert` and probably `rs_bag2image` can do is to extract the frame saved in the bag file. 

That is to say, if you found out that your RGB and depth frames are not synchronized, it could the **recording** process that failed to synchronize streams from different sensors.

Dropping frame issues are common in D435 when I did the experiments, and it makes the time synchronization a little bit tricky. I only worked with depth and IR frames. They all come from the same sensor, so it is easy to get synchronization using timestamps and frame numbers. In your case, my suggestion is to make use of the hardware trigger, timestamps and the frame number API from RealSense SDK, or accurate external timer for getting the exact timing information. 

I hope that helps.

Yang

ankith001, 2019-04-30:
Hi @YangJiao1996, 

Thank you for your response. Any easy way to get the timestamps and frame number from the .bag file?
Have you come across any code which does that?

YangJiao1996, 2019-04-30:
Hi @ankith001,

Please go through the API or example code for getting timestamps and/or frame numbers. Una's code is a great example to get started with. 

One problem with using these to synchronize the streams is that both the timestamps and frame numbers are accumulated (from the time when sensors are activated), and you need to be aware of that.

Yang

closed 2019-05-29 by RealSenseCustomerSupport