realsenseai/librealsense #3691: cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/lib/python2.7

liangzhicong456, 2019-04-06:
CMake Error at wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
  Python config failure:

Call Stack (most recent call first):
  wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/librealsense/build/CMakeFiles/CMakeOutput.log".

PS: FindPythonLibsNew.cmake:95 (message):
if(PYTHONLIBS_FOUND AND PYTHON_MODULE_EXTENSION)
    return()
endif()

# Use the Python interpreter to find the libs.
if(PythonLibsNew_FIND_REQUIRED)
    find_package(PythonInterp ${PythonLibsNew_FIND_VERSION} REQUIRED)
else()
    find_package(PythonInterp ${PythonLibsNew_FIND_VERSION})
endif()

if(NOT PYTHONINTERP_FOUND)
    set(PYTHONLIBS_FOUND FALSE)
    return()
endif()

# According to http://stackoverflow.com/questions/646518/python-how-to-detect-debug-interpreter
# testing whether sys has the gettotalrefcount function is a reliable, cross-platform
# way to detect a CPython debug interpreter.
#
# The library suffix is from the config var LDVERSION sometimes, otherwise
# VERSION. VERSION will typically be like "2.7" on unix, and "27" on windows.
execute_process(COMMAND "${PYTHON_EXECUTABLE}" "-c"
    "from distutils import sysconfig as s;import sys;import struct;
print('.'.join(str(v) for v in sys.version_info));
print(sys.prefix);
print(s.get_python_inc(plat_specific=True));
print(s.get_python_lib(plat_specific=True));
print(s.get_config_var('SO'));
print(hasattr(sys, 'gettotalrefcount')+0);
print(struct.calcsize('@P'));
print(s.get_config_var('LDVERSION') or s.get_config_var('VERSION'));
print(s.get_config_var('LIBDIR') or '');
print(s.get_config_var('MULTIARCH') or '');
"
    RESULT_VARIABLE _PYTHON_SUCCESS
    OUTPUT_VARIABLE _PYTHON_VALUES
    ERROR_VARIABLE _PYTHON_ERROR_VALUE)

if(NOT _PYTHON_SUCCESS MATCHES 0)
    if(PythonLibsNew_FIND_REQUIRED)
        message(FATAL_ERROR
            "Python config failure:\n${_PYTHON_ERROR_VALUE}")
    endif()
    set(PYTHONLIBS_FOUND FALSE)
    return()
endif()

I guess this is python not configured, but i have no idea to handle, hope someone can help me, Thank you very much!

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi liangzhicong456,

To help us better understand and support, can you please share more information about the issue?
What is your build system (which nvidia platform, and what version of L4T for nvidia)? Did you encounter any issue when following instructions here:
a) https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
b) https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

Thanks!

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

Hi liangzhicong456,

Wonder if there is any update? Or do you still need help?

Thanks!

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

Hi liangzhicong456,

If there is nothing else needs help, this one will be closed soon.

Thanks!

closed 2019-05-13 by RealSenseCustomerSupport