realsenseai/librealsense #3692: Does  the librealsense team support installation in Ubuntu on Windows 10?

coldcoda, 2019-04-06:
Hello @dorodnic , @RealSense-Customer-Engineering , @seebq ,
Does  the librealsense team support installation in Ubuntu on Windows 10?

coldcoda, 2019-04-06:
Here is the problem I met! Is it a fatal problem?
[image]

dorodnic, 2019-04-07:
> Does the librealsense team support installation in Ubuntu on Windows 10?

Officially, no. But it should be possible and we are likely to look into it in the future.

> Here is the problem I met! Is it a fatal problem?

No, I think you can skip this step.

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@coldcoda  If no other questions for this ticket, will close it accordingly. Thanks!

closed 2019-05-12 by RealSenseCustomerSupport