realsenseai/librealsense #3693: How can I solve it When I build from sources?

coldcoda, 2019-04-06:
When I build with the Cmake ,here is the download error from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.5448/target-0.0.18.5448.mvcmd. And how to solve it? 
[image]

coldcoda, 2019-04-07:
Hello @dorodnic , @matkatz 
And when I accessing this link directly from the browser, It can be download correctly. So what can I do to solve it?
Thanks a lot!!

dorodnic, 2019-04-07:
https://github.com/IntelRealSense/librealsense/issues/3182#issuecomment-459493048

coldcoda, 2019-04-07:
Hello @dorodnic ,
Because I build it at Ubuntu on windows 10, so I really don't know how to disable the BUILD_WITH_TM2 cmake flag.
Cause I don't have a GUI to check the BUILD_WITH_TM2 or not. So what can I do to disable it?
Thanks!!

dorodnic, 2019-04-07:
I assume you are running `cmake ..` in the terminal. Please try `cmake .. -DBUILD_WITH_TM2=false`

coldcoda, 2019-04-07:
Hello @dorodnic ,
I have solved the problem.
Thanks a lot!! 
So does the BUILD_WITH_TM2 is prepared for RealSense T265? Because I use D435, so I don't need it?

dorodnic, 2019-04-07:
Right :)

coldcoda, 2019-04-07:
Hello @dorodnic ,
Sorry for that When I run the rs-capture, I met a new problem again.(Could not open OpenGL window, please check your graphic drivers or use the textual SDK tools)
[image]
I tried to solve it by myself, but I don't solve it.
So what can I do to solve it?
Thank you!!

dorodnic, 2019-04-07:
Hi @coldcoda 
I'm not familiar at all with Linux subsystem on Windows, but it is not surprising that graphical examples will be problematic to run. This may be possible, but likely to require more steps, since you are running in a textual session. 
How about `rs-save-to-disk` and other command line tools?

coldcoda, 2019-04-07:
I think it doesn't run correctly. When I run rs-save-to-disk it just shows that (RealSense error calling rs2_pipeline_start(pipe:0x7fffd7b180f0):
    No device connected)
[image]
Have I missed some steps when I build the sources?

coldcoda, 2019-04-07:
Hello @dorodnic ,
I think it doesn't run correctly. When I run rs-save-to-disk it just shows that (RealSense error calling rs2_pipeline_start(pipe:0x7fffd7b180f0):
No device connected)
[image]

Have I missed some steps when I build the sources?

dorodnic, 2019-04-07:
Check if `lsusb` is available to you and what is its output. 
Again, we do not officially support this configuration, so I cannot guaranty it will work.

coldcoda, 2019-04-07:
Hello @dorodnic 
The lsusb doesn't have any output.
[image]
So what's the matter of it? Can I solve it?

ev-mp, 2019-04-07:
@coldcoda , I've checked the [MS support for Ubuntu site](https://docs.microsoft.com/en-us/windows/wsl/faq) and it appears that the features that are currently supported are bash-like terminal tools that have access the MS file system.  
There is no mention of HW drivers required to use USB/UVC devices, nor OpenGl graphic drivers - only terminal.

So IMO, the only viable use-case to run Librealsense with Ubuntu on Windows is for exercising playbacks from pre-recorded rosbags using command prompt as an output, e.g. visualizing depth data in `rs-depth` character-map style.

coldcoda, 2019-04-07:
Hello @ev-mp , @dorodnic 
Thanks a lot! I think I know it.

coldcoda, 2019-04-07:
Hello @dorodnic  , @ev-mp 
After I complete build from the source, and when I try to run the example of pcl . I have met the error.
[image]
Should I need to install glfw extra? Or how solve it?

fqw1727252209, 2019-04-14:
> Hello @dorodnic , @matkatz
> And when I accessing this link directly from the browser, It can be download correctly. So what can I do to solve it?
> Thanks a lot!!

you can try "sudo apt update",i use this method,it works well!

coldcoda, 2019-04-18:
Thanks a lot!!

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@coldcoda If no other questions about this ticket, will close it accordingly. Thanks!

closed 2019-05-12 by RealSenseCustomerSupport