realsenseai/librealsense #3694: rs-convert not using preset

hueri, 2019-04-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  v2.19.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi there,

I'm a semi-non-developer, trying to record some color and depth for video purposes.  I've been able to tune the camera to the specs that i want, save a preset, and record a bag.  

The issue I'm having is that when I load the recorded sequence into the Realsense Viewer, it matches the Preset's settings, but when i use rs-convert.exe, it uses the default settings.  Not being a C++ / C# developer, i don't see an easy way to add a flag for a Preset.json file in rs-convert.  Is there another way to do that, or could that feature be added?  

Please and thanks!

.m

dorodnic, 2019-04-07:
Hi @hueri 
Currently only settings affecting the hardware are saved into the JSON, and these will be reflected in recorded data.
Software configuration, such as post-processing and depth visualization settings, are currently not being saved, but we are planning to add them in the future.

hueri, 2019-04-07:
Thanks for responding!  
If i'm understanding correctly, for example the only way to export png's with white to black value ranges is to modify and compile something like a version of rs-convertWB?  

Thanks much!

dorodnic, 2019-04-08:
You are correct. However, doing this in python will most likely be much faster. 
Check out [read_bag_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py) to get started. 
To modify color scheme you need to `set_option(rs.option.color_scheme, ...)` on the `colorizer` object

hueri, 2019-04-09:
Ah, cool - thanks!  
Will definitely need to look into that.  Cheers!

closed 2019-05-15 by RealSenseCustomerSupport