realsenseai/librealsense #3696: rs2_load_json() error

TouchDeeper, 2019-04-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.18.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
The core code is below:
```
    //load customized preset
    auto sensor2 = profile.get_device();
    std::string jsonPath = "./ShortRangePreset.json";
    auto advancedDepthSensor = sensor2.as<rs400::advanced_mode>();
    advancedDepthSensor.load_json(jsonPath);
```
The error is below:
```
RealSense error calling rs2_load_json(dev:0x14a9710, json_content:0x143af60, content_size:23):
    parse error - unexpected '.'
```
I read the related issue. It seems like has been fixed, but I encounter the error again.
Can you provide some advice? Thanks a lot!

dorodnic, 2019-04-08:
Seems like the problem is in the JSON file content. Please check you are passing valid JSON file

TouchDeeper, 2019-04-08:
Hello @dorodnic 
I check the json file, it is downloaded from the realsense wiki. I think the json file is no problem.
The error may caused by the load_json function, because when I modify the `std::string jsonPath = "./ShortRangePreset.json";` to `std::string jsonPath = "/ShortRangePreset.json";`, the error will be:
```
RealSense error calling rs2_load_json(dev:0x14a9710, json_content:0x143af60, content_size:23):
    parse error - unexpected '/'
```

dorodnic, 2019-04-08:
Oh... Sorry. Load JSON is expecting file **content**, not file name.

TouchDeeper, 2019-04-08:
Oh, so how can I load all the parameters of a .json file?

dorodnic, 2019-04-10:
```cpp
std::ifstream file("ShortRangePreset.json");
std::string str((std::istreambuf_iterator<char>(file)), std::istreambuf_iterator<char>());
advanced.load_json(str);
```

TouchDeeper, 2019-04-11:
Hello @dorodnic 
It works, thanks a lot!

closed 2019-04-11 by TouchDeeper