realsenseai/librealsense #3697: T265 connection/recognition issues

eospi, 2019-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 and T265 } | 
| Firmware Version                   |  05.11.01.100/0.0.18.5448 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2.20                           | 
| Language                            |   Python and C++                      | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm trying to use the D435 and T265 together. In Python, pyrealsense can detect and use the D435 but not the T265. For example, when I run the following code, only the D435 is recognized. 
```
context = rs.context()
for device in context.query_devices():
    print(device)
```

I get this error:
Device-4D00: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)

This issue has rotated through Python, C++, and the RealSense Viewer with the same error, and it can work in one environment but not the others. How can I fix this? Thanks!

ev-mp, 2019-04-11:
@eospi , I could reproduce this error on a PC that has no proper Librealsense `udev-rules` installed.
Please verify `udev-rules` are installed with:
1.   [the installation script](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#prerequisites) when building from source code:
>./scripts/setup_udev_rules.sh 

2. Via [debian distribution](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) for Ubuntu LTS:
>sudo apt install librealsense2-udev-rules

 Then reconnect T265 and retry.

eospi, 2019-04-11:
Thanks for the response! I installed `udev-rules`, and it worked the first time I tried it. However, it has reverted to the previous issue, and the T265 will not appear in the RealSense Viewer.

eospi, 2019-04-13:
The T265 seems to be connecting most of the time, but sometimes it won't be recognized with the same libusb error. When recognized, running `t265_example.py` works as expected, but when I use both the T265 and D435 together, the pose data from the T265 is empty. Trying to access `data.translation` causes an Attribute Error:
`'pyrealsense2.pyrealsense2.BufData' object has no attribute 'translation'`

ev-mp, 2019-04-14:
@eospi , 
I'd suggest to run `rs-multicam` example to see whether the D435/T265 tandem works properly.
then you can may post the python code segment for review.
Note that currently there are limitations imposed on how T265 shall be activated in multi-cam scenario. (#3734).

eospi, 2019-04-15:
rs-multicam works well. Here is my code:
```
import pyrealsense2 as rs
import numpy as np

context = rs.context()
pipelines = []

for device in context.query_devices():
    pipe = rs.pipeline(context)
    config = rs.config()
    config.enable_device(device.get_info(rs.camera_info.serial_number))
    pipe.start(config)
    pipelines.append(pipe)
    print(device)

def grab_colour_frame():
    frames = pipelines[0].wait_for_frames()
    colour = np.asanyarray(frames.get_color_frame().get_data())
    return colour

def grab_pose_frame():
    frames = pipelines[1].wait_for_frames()
    pose = frames.get_pose_frame()
    return pose

while True:
    colour = grab_colour_frame()
    pose = grab_pose_frame()
   
    if pose:
        data = pose.get_data()
        print(data.translation)
```

ev-mp, 2019-04-15:
Small fix will make the code run properly:
>if pose:
        data = pose.~~get_data~~**get_pose_data**()
        print(data.translation)

See [reference example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_example.py#L32)

eospi, 2019-04-16:
Thanks so much! It's working now.

closed 2019-04-16 by eospi

dhirajpatnaik16297, 2019-12-30:
Hey,
I am getting the following error while i am connecting both d435 and t265 to jetson xavier
frames = pipelines[0].wait_for_frames()
IndexError: list index out of range

but, individually they are working and in realsense-viewer both are detected and working fine.
Only in the python code, i am not able to make it work.
Please help me out.