realsenseai/librealsense #3698: The D435 set color exposure will effect the ir exposure?

Gitlaber, 2019-04-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } |  
| Operating System & Version   |   {ubuntu16.04  |                                          |
| Platform                                 | PC  |
| SDK Version                           |2.17 |                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
We want to set deduce the color camera's exposure and do that, but we find one of the D435 ir camera also be changed exposure when we deduce the color. Another is work correctly, anyone had met this question? It is the camera is broken?

Using the same code as it:
`	auto colorSensors = profile.get_device().query_sensors()[1];
	auto irSensor = profile.get_device().query_sensors()[0];

	if(colorSensors.supports(RS2_CAMERA_INFO_NAME)) Log_Info << colorSensors.get_info(RS2_CAMERA_INFO_NAME);
    if(irSensor.supports(RS2_CAMERA_INFO_NAME)) Log_Info << irSensor.get_info(RS2_CAMERA_INFO_NAME);

    colorSensors.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0); //zero is turn off
    colorSensors.set_option(RS2_OPTION_EXPOSURE, 50);
    colorSensors.set_option(RS2_OPTION_GAIN, 64);
	irSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
	irSensor.set_option(RS2_OPTION_LASER_POWER, 360);`

The bad one:
[image]

The good one:
[image]

dorodnic, 2019-04-08:
The two sensors should not be influencing each other. 
Did you try the same experiment in the RealSense Viewer?
Are the cameras using the same firmware version?

Gitlaber, 2019-04-08:
@dorodnic it can work well both of them in the RealSense Viewer. But the cameras have different firmware version.the good one version is 05.08.15.00 and the bad one is 05.10.06.00. Should i update the firmware version both of them?

Gitlaber, 2019-04-09:
Okay... I revert the bad one firmware to the 05.08 that is can work normally... So what does the meaning of update? It really make a serious bug. @dorodnic

Gitlaber, 2019-04-11:
@doronhiI was tested the SDK2.19 with the firmware version 05.11.100 and i found that the ir image will twinkle when i startup the program.I really don't konw where i was wrong and it confuse me for few day.We use the same code all the time. I really need your help right now.

Gitlaber, 2019-04-11:
@doronhi   I was tested the SDK2.19 with the firmware version 05.11.100 and i found that the ir image will twinkle when i startup the program.I really don't konw where i was wrong and it confuse me for few day.We use the same code all the time. I really need your help right now.

Gitlaber, 2019-04-11:
@dorodnic  By the way, it can work well in the Realsesen-Viewer.Is my code wrong?

RealSenseCustomerSupport, 2019-04-17:
----------------------------------------------

Hi @Gitlaber,

To clarify:
What's the final firmware version you used for both D435 cameras? No matter which firmware version, you mean realsense-viewer works normally on both cameras?

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Not see any issue about your partial code.  I don't observe this issue.

Gitlaber, 2019-04-19:
> Not see any issue about your partial code. I don't observe this issue.

I found the firmware version down 05.10 that both cameras can work normally. If the firmware above 05.10 change the color exposure will effect the ir.

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

@Gitlaber Suggest to modify below two lines. The sensor order might be not reliable. You can refer to the code snippet in https://github.com/IntelRealSense/librealsense/issues/2637 to check the sensor type and then configure the exposure.

 auto colorSensors = profile.get_device().query_sensors()[1];
auto irSensor = profile.get_device().query_sensors()[0];

Gitlaber, 2019-04-30:
@RealSenseCustomerSupport Okay... But I personally think it must be the firmware version's problem. Because I use the same coding :

```
        auto colorSensor = profile.get_device().query_sensors()[1];
	auto irSensor = profile.get_device().query_sensors()[0];
        std::cout << colorSensors.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
        std::cout << irSensor.get_info(RS2_CAMERA_INFO_NAME) << std::endl ;
        colorSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
        colorSensors.set_option(RS2_OPTION_EXPOSURE, 150);
        colorSensors.set_option(RS2_OPTION_GAIN, 64);
        irSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
        irSensor.set_option(RS2_OPTION_LASER_POWER, 360);
```

"The bad one" and "The good one" both can log the their information. After I downgraded "The bad one"  to the 05.08.15.00, it can be used normally.

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@Gitlaber  I tried your code snippet on win10 and Ubuntu16.04 with librealsense v2.21.0 but sorry that I couldn't reproduce your issue with FW 5.10.6 and 5.11.4. The exposure control of RGB and IR are independent seen from my result.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@Gitlaber  Any update from your side with the modifications about sensor order? Did you still get the issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@Gitlaber  Will close the ticket if you have no other questions. Thanks!

jkflying, 2019-05-27:
Have you tested both USB2 and USB3? We've seen some issues that show on USB2 but not USB3 regarding exposure on 5.10.x firmware

closed 2019-05-28 by RealSenseCustomerSupport