realsenseai/librealsense #3699: When I run the example of PCL, I met the error

coldcoda, 2019-04-08:
After complete build from source, and try to run the example of PCL. I met the error. It can't  run.(With VS2017 x64, and PCL msvc2017-win64)
[image]
Can you give me some advice?
Thanks a lot!

coldcoda, 2019-04-08:
Hello @dorodnic , @ev-mp 
After complete build from source, and try to run the example of PCL. I met the error. It can't run.(With VS2017 x64, and PCL msvc2017-win64)
[image]

Can you give me some advice?
Thanks a lot!

vladsterz, 2019-04-08:
Hello @coldcoda, this problem occurs when your program cannot find the .dll file specified. To resolve it you have to find where PCL is installed, locate this specific .dll file and copy-paste it to the location of the program you try to execute.

coldcoda, 2019-04-18:
Thanks a lot! And I have solved it!

closed 2019-04-25 by RealSenseCustomerSupport