realsenseai/librealsense #3700: Cmake  version2.20 have some issues

zhuoyueljl, 2019-04-08:
-- - Downloading FW 0.0.18.5448 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.5448/target-0.0.18.5448.mvcmd
--   Converting FW version 0.0.18.5448 from target.mvcmd to /home/cver/Topband/T265/2.20/librealsense-master/third-party/libtm/libtm/src/fw.h
--   File /home/cver/Topband/T265/2.20/librealsense-master/third-party/libtm/libtm/src/fw.h wasn't changed
-- - Downloading Central App 2.0.19.271 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/app/2.0.19.271/central_app-2.0.19.271.bin
--   Converting Central App version 2.0.19.271 from central_app.bin to /home/cver/Topband/T265/2.20/librealsense-master/third-party/libtm/libtm/src/CentralAppFw.h
--   File /home/cver/Topband/T265/2.20/librealsense-master/third-party/libtm/libtm/src/CentralAppFw.h wasn't changed
-- - Downloading Central BL 1.0.1.112 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/bl/1.0.1.112/central_bl-1.0.1.112.bin
--   Converting Central BL version 1.0.1.112 from central_bl.bin to /home/cver/Topband/T265/2.20/librealsense-master/third-party/libtm/libtm/src/CentralBlFw.h
--   File /home/cver/Topband/T265/2.20/librealsense-master/third-party/libtm/libtm/src/CentralBlFw.h wasn't changed

zhuoyueljl, 2019-04-10:
update the code again, it works well!

closed 2019-04-10 by zhuoyueljl

fqw1727252209, 2019-04-14:
You can try "cmake BUILD_WITH_TM2=FALSE ../"

fqw1727252209, 2019-04-14:
> update the code again, it works well!

work well