realsenseai/librealsense #3701: rs-distance two camera connection

sophie-shin, 2019-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Jetson Tx2 Ubuntu  | 

### Issue Description
<Request>

Hello.

Regarding  " rs-distance " , I want to connect two realsense cameras.(two D435i)
Two D435i cameras were configured in correctly since it shows 
```
nvidia@tegra-ubuntu:/usr/local/bin$ ./rs-distance
There are 2 connected RealSense devices.
```
When the cameras are connected, 'rs-distance' result is from the camera linked with front part of usb hub.

So now I need to change some code to get the distance data from two camera simultaneously.
Now i am working on to cope all of .cpp and .h and edit file name as "rs_frame2" "rs_option2" ... to distinguish two cameras information.

But I am worry to make a mistake while I edit a code in detail.

Is there any way to get two cameras 'rs-distance' results in one screen?

many thanks

```
printf("The camera1 is facing an object %.3f meters away.\n", dist_to_center);
printf("The camera2 is facing an object %.3f meters away.\n", dist_to_center2);
```

AntoineWefit, 2019-04-26:
I guess you could try to start 1 pipeline for each camera, both of them initialized with unique serial number.
You'll then need to implement some kind of multi-threading to run the cameras simultaneously.
I'm far from being an expert in IT, but what I'm suggesting is working for a similar application : maybe #2219 could help you ?

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

Hi sophie-shin,

Wonder if there is any update on this one? What AntoineWefit suggested would be a good approach.

Thanks!

sophie-shin, 2019-05-07:
Sorry, I am still struggling about this.

AntoineWefit, 2019-05-13:
I assume you're working on [this](https://github.com/IntelRealSense/librealsense/blob/bcc71f3b01c482dfa51895ab6247135c9c76323d/examples/C/distance/rs-distance.c).
I don't know much about C, so here is my guess : 

Here you are retrieving 1 device with `dev` : 
```
    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
rs2_device* dev = rs2_create_device(device_list, 0, &e);
```
Try declaring 2 instead (I think that it's usually "safer" to select devices with serial numbers instead of their ID but this should also work) : 
```
rs2_device* dev1 = rs2_create_device(device_list, 0, &e);
rs2_device* dev2 = rs2_create_device(device_list, 1, &e);
```

Same thing for the pipeline : 
```
    // Create a pipeline to configure, start and stop camera streaming
    // The returned object should be released with rs2_delete_pipeline(...)
rs2_pipeline* pipeline = rs2_create_pipeline(ctx, &e);
```
Make it 2 : 
```
rs2_pipeline* pipeline1 = rs2_create_pipeline(ctx, &e);
rs2_pipeline* pipeline2 = rs2_create_pipeline(ctx, &e);
```

Now the tricky part : 
```
    // Start the pipeline streaming
    // The retunred object should be released with rs2_delete_pipeline_profile(...)
rs2_pipeline_profile* pipeline_profile = rs2_pipeline_start_with_config(pipeline, config, &e);
```
I don't know how to start 2 pipelines in C & using multi-threading to handle them both simultaneously (if possible), you might want to ask dorodnic about that.

I could help you a bit if you do this in C++ though.

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@sophie-shin Did you get it through as AntoineWefit's suggestion? And you can also refer to the multicam example https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam to see if it can help you also.

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@sophie-shin Any other questions for this ticket?

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@sophie-shin Any good news from your side? Looking forward to your reply. Thanks!

closed 2019-06-28 by RealSenseCustomerSupport