realsenseai/librealsense #3703: Drivers with Camera streaming from D435i for MacOS

pzoltowski, 2019-04-08:
Would it possible to release drivers that support only video feed for now (without motion sensor) for Mac OS and/or Android? This should be easier to do I guess (since d430 is already supported) and then add support for HID motion sensor later. I'm considering buying D435i but it's still not supported for Mac OS. I cannot dual boot because I'm using soft raid 0 which is not supported by bootcamp. I would be happy with video feed for now but don't want to buy d430 just to sell it couple of months later and then buying d435i. Some ETA of drivers supporting other system would be great anyway.

bobgates, 2019-04-12:
I'd like to second this. Since 2.17.0 the site has said "Windows 7 and Mac OS support is expected to be added in future releases." Please can Mac OS support be added? Like pzoltowski I could live with the 435 driver and add motion support later, but I'd really like to get the vision system running on my Mac.

pierrextardif, 2019-05-28:
I'd be glad to have it too.
As mentionned [here](https://github.com/IntelRealSense/librealsense/issues/3237#issuecomment-487725834), processing does have a library which works with a D435i. but only the color stream though, :/ ...
If you want to do some quick test...

dorodnic, 2019-05-29:
Hi @pzoltowski @bobgates @pierrextardif 
I cannot test it right now, but looking at the code, with latest version on Mac OS librealsense should detect D435i and provide depth and color information - https://github.com/IntelRealSense/librealsense/pull/3804
I apologise for the delay in motion support. Unlike other features that wait in the backlog to be prioritised, this feature was planned, is being worked on for many months now and getting delayed due to complications in implementation. Maybe we chose the wrong approach, but when it is ultimately complete we are going to have a cross-platform open-source user-space driver for HID sensor devices, which can bring significant value not just for RealSense but to open-source community in general (somewhat similar to what libuvc did for UVC devices)
HID sensor drivers are far from being ubiquitous, both Linux kernel module and Windows implementation have many issues in them, so we had to rebuild a lot of the protocol from scratch using only the Spec and USB analyser. Fortunately, it's getting close to finished.

pierrextardif, 2019-05-29:
Hi @dorodnic 
Thanks, I have tried to test it, with no luck.
There is a step in the cmake which doesn't work, and isn't listed in the errors so far, but I'm not sure where it comes from.
However, the ``` brew install librealsense ``` as mentionned [#2811](https://github.com/IntelRealSense/librealsense/issues/2811) installed it for me. ( further steps did not work, either with the latest develop branch).
My goal was to use it with Openframeworks, so I updated [my version](https://github.com/pierrextardif/ofxRealsense2/) of [the repo](https://github.com/hiroMTB/ofxRealsense2) from hiroMTB, and I works!
Essentially 2.2.22 works with D435i for me.
++

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@pierrextardif So you got it through with SDK v2.22.0, right? Any other questions about this ticket? Thanks!

bobgates, 2019-06-13:
Yeah, I'm happy.

pierrextardif, 2019-06-13:
yes, that's fine thanks for the help.

Best,

P

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@pierrextardif @bobgates  Great! Thanks for the update!

closed 2019-06-13 by RealSenseCustomerSupport