realsenseai/librealsense #3704: Depth image seems to be bad

SamConti, 2019-04-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.38  | 
| Platform                                 | Raspberry Pi  |
| Language                            | python | 

### Issue Description
Hello,

I try to retrieve image from D435 camera in python but I think my depth image is not good and I don't know why.

Here the image :

[image]

And the depth image :

[image]

My code is 

`import pyrealsense2 as rs`
`pipeline = rs.pipeline()`
`config=rs.config`
`config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16,6)`
`config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8,6)`
`config.enable_record_to_file("bagfiletest.bag")`
`pipeline.start()`
`pipeline.wait_for_frames()`
`pipeline.stop()`

dorodnic, 2019-04-08:
Yes, the depth frame is bad.
Did you try skipping couple first frames? Auto-expose needs some time to stabilize. 
If this persists, it is possible there is a problem with camera calibration

SamConti, 2019-04-08:
Okay, it seems it needs some time to stabilize, but it's a problem.
My final using is to have a bag file each 5mn with one image inside and I send it on a ftp server, if I have to wait for 5s before stopping, bag file is too big (>20Mo) and I have several images into the bag file.
Is there a possibility to take one clean image ?

dorodnic, 2019-04-11:
You could try `save_single_frameset` defined in `rs_export.hpp` to manually save one specific frame-set.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@SamConti Could you please shed more details about what kind of python document you need? For python, we have some examples in https://github.com/IntelRealSense/librealsense/tree/v2.21.0/wrappers/python/examples Maybe you can find more information there. Thanks!

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@SamConti Any other questions for this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@SamConti If no other questions, will close this ticket accordingly. Please feel free to create new ticket if have other questions. Thanks!

closed 2019-05-16 by RealSenseCustomerSupport

soarwing52, 2019-05-23:
@dorodnic 
I tried the save_single_frameset as you mentioned, I can get proper visualized depth image
but when I want to get the depth data, the deproject, there's always only 0.00
 #4020 this is my issue