realsenseai/librealsense #3707: Is librealsense v1.x actively maintained?

kjkjava, 2019-04-08:
Is librealsense v1.x still being actively maintained/supported?

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md describes how v2.x will drop support for certain devices but v1.x will still be maintained. However, I don't see a 1.x branch with bug fixes, etc. It looks like v1.x devices are all considered legacy now, right? (For what it's worth, I don't need 1.x; just hoping to get a better understanding of the state of that code.) Thanks!

MartyG-RealSense, 2019-04-08:
I believe Legacy Librealsense is still in use for older RealSense cameras such as F200 and R200 that the current SDK does not support, but it has not been updated for a couple of years.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

There is a user created 'fork' of it that was updated as recently as 5 months ago, but Legacy Librealsense is not receiving any more official updates from Intel.

https://github.com/billyzs/librealsense-legacy

kjkjava, 2019-04-08:
Good to know. Thank you!

closed 2019-04-08 by kjkjava