realsenseai/librealsense #3708: Realsense - dropping frames

rabysam28, 2019-04-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-46-generic | 
| Platform                                 | DELL  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C++11 | 
| Segment			|  other }                          | 

### Issue Description
I am recording BAG file using realsense-viewer, and after that I am converting them to PNG files, and I see the following issues:
1) Although the freq is set to 30Hz  realsense-viewer, the number of PNG files are less than 30 images per seconds. I am loosing some frames.
2) I noticed that there are cases that two consecutive frames has the same content, meaning that the two PNG files are showing exactly the same thing, but they _do_ have two different timestamps.
3) There are also cases where two consecutive frames are two different things which is good, but they have the exact same timestamp. (This is opposite of the issue 2)
Could you please advise on the above three issues?
Thanks

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Hi rabysam28,

Can you try the built-in rosbag-inspector tool on your recorded bag file, and see if the inspector reports same issues?
Also wonder when recording to the bag file, how many streams and what resolution of each stream?

Thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi rabysam28,

Wonder if there is any update? Or do you still need help on this one?

Thanks!

closed 2019-05-17 by RealSenseCustomerSupport