realsenseai/librealsense #3709: Relocalize to imported map

JBBee, 2019-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5448 | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  2.20.0 }                          | 
| Language                            |  Unity/C++                          | 
| Segment			|  AR                          | 

### Issue Description
Thanks for the latest release, 2.20.0 and for all your help so far.  We're working with the latest to use in our persistent AR application.  We're seeing some localization issues and trying to characterize them and see if there's any advice you can provide on how to improve the process of re-localizing to an imported map.  

We first create a map and set a static node called 'origin' to (0,0,0).  We let the mapping process go for roughly two minutes while walking all around the room.  Then we export and save the map.  Then we startup the process again and import the map and see how soon we can fetch the 'origin' static node.  We can fetch it within a few seconds when we start the T265 in the same location and orientation as when the initial map was started.  But if we start in a location that's not exactly the original starting position we can never fetch the origin until we return to that original location.  Is there anything we can do to improve upon this when making the map?  Does it matter how soon after starting the initial map that we create the static node?  Would you recommend we create the static node early in the mapping process or late?  Any advice on how to allow the user to re-localize to the map from anywhere in the room would be great.

Thanks.

Filmengineers, 2019-04-16:
Yes. Exactly what I'm looking for too.

giovannicairoli, 2019-04-18:
Hi @JBBee,

Could you please share with me how to create and save a map? I'm super inexpert and I'm trying to find documentation.

Thank you

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

We are working on an updated version of the relocalization feature.  Track the upcoming releases of librealsense.  I will also comment here when the updated version of relocalization is released in the package.

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Version 2.21.0 is available...Next version is pending.  As stated we are always making improvements to our algos including relocalization.

There are calls to import and export relocalization maps within the librealsense src

besimpler, 2019-05-27:
Is it possible to import/exmport map inside standard application that you provide?

RealSenseCustomerSupport, 2019-06-01:
----------------------------------------------

At this time we do not have a sample that outlines import/export maps.  We do not have a timeframe right now as to when a future release will have example on mapping function..  Continue to monitor future releases for updates, new features and new sample additions.

closed 2019-06-01 by RealSenseCustomerSupport

JimmyFroost, 2019-06-26:
Hi,

Any update available on this matter ?