realsenseai/librealsense #3710: Confused Loading Presets in C++

Alpatzino, 2019-04-08:
| Required Info                  
| Camera Model                       |  D400 | 
| Firmware Version                   | (2.18.0) | 
| Operating System & Version |   Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  (e.g. 4.4.0-143generic)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++}                          | 
 

Hello,

I am trying to fetch data from the depth sensor with 640x480 resolution, 60 FPS and of the presets activated. I am pretty confused by all the different ways  to set the preset. I want to load the preset in C++ API. I use the following code: 
`//

    //Construct a pipeline which abstracts the device
    rs2::pipeline pipe;
    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    //Add desired streams to configuration
    
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 60);

    //Instruct pipeline to start streaming with the requested configuration
    auto profile = pipe.start(cfg);
    auto sensor = profile.get_device().first<rs2::depth_sensor>();
	sensor.set_option(rs2_option::RS2_OPTION_VISUAL_PRESET, rs2_rs400_visual_preset::RS2_RS400_VISUAL_PRESET_HAND);`
Will this work or do I have to activate advanced mode first? Is there a better way to load the preset? 
And will it interfere with the fps and resolution?

Many thanks in advance

dorodnic, 2019-04-10:
Hi @Alpatzino 
Before using any presets, Advanced Mode must be enabled, **once**. 
After that, doing `sensor.set_option(RS2_OPTION_VISUAL_PRESET, x)` and `dev.as<advanced_mode>().load_json()` will produce the same results. 
Both calls will not augment FPS and resolution and only apply the relevant controls.

Alpatzino, 2019-04-11:
Thank you, I think I will get this to work. Best of luck :)

closed 2019-04-22 by RealSenseCustomerSupport