realsenseai/librealsense #3714: Raspberry Pi 3B+ with T265

sirisian, 2019-04-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | Segfaults randomly before I can access | 
| Operating System & Version |   Ubuntu 16.04.6 (Ubuntu Mate)  | 
| Kernel Version (Linux Only)    |  4.19.20-v7+                                         | 
| Platform                                 | Raspberry Pi 3B+ |
| SDK Version                            |  2.20.0                         | 
| Language                            |  None                          | 
| Segment			|                            | 

### Issue Description

I have a Raspberry Pi 3 B+. I was trying to make a mobile T265 unit to walk around with. I followed these steps and everything built fine: https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

The realsense-viewer immediately segmentation faults when I run it. One time it opened and I could interact with it, but clicking them did nothing. When I plugged in the T265 camera it segmentation faulted again, and I could never get it to open again.

When I run the rs-pose and such it just says "No device connected", yet dmesg shows the usb messages for the "Product: Movidius MA2X5X" being plugged into the USB when I unplug and plug it in.

There's a new 18.04.2 Ubuntu Mate. Would it be possible to support that? Most images don't seem to work with the 3 B+, but that one does. Or does it already support it? What do the kernel patches do for the older kernels?

Has anyone tested trying to get the T265 working with a Raspberry Pi 3 B+? Just as a sanity check, I'm not running into any power issues plugging it into the Pi's USB 2.0 port?

dlarue, 2019-04-12:
it'd got it working and should work with the raspbian OS's since I used raspbian lite to test. See this thread - https://github.com/IntelRealSense/librealsense/issues/3517

sirisian, 2019-04-14:
@dlarue I got it to work on the Ubuntu Mate 18.04.2 just building the source with no changes.

Found one small issue, but I'll open a new issue.

closed 2019-04-14 by sirisian

dlarue, 2020-01-20:
If anyone finds this, here's what worked today on Raspbian Buster-lite:

Raspberry Pi OS: Raspbian Buster-lite dated 2019-09
Raspberry Pi Hwr:  Raspberry Pi 4 2GB version(no swap added)

#~/bin/bash

# built on rPi4 2GB with no added swapfile
cd
sudo apt update
sudo apt-get install -y cmake python3-dev raspberrypi-kernel-headers libusb-1.0-0-dev
git clone https://github.com/IntelRealSense/librealsense
cd librealsense/
git checkout master
git pull
mkdir build
cd build/
cmake -D CMAKE_BUILD_TYPE="Release" -D FORCE_LIBUVC=ON -D BUILD_PYTHON_BINDINGS=ON -D BUILD_EXAMPLES=ON -D BUILD_GRAPHICAL_EXAMPLES=OFF -D BUILD_EASYLOGGINGPP=OFF -D BUILD_GLSL_EXTENSIONS=OFF ..
make -j3