realsenseai/librealsense #3717: Read auto-exposure values

SamConti, 2019-04-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.38  | 
| Platform                                 | Raspberry Pi  |
| Language                            | python | 

### Issue Description
Hello,

I would like to know if it's possible to read the values set by auto-exposure ?
(I don’t even know what values are set up by auto-exposure).

In fact I would like to read these values then to set-up them manually to avoid auto-exposure time in my bag file.

My code algorithm shoudl be :

- Record a bag file with auto-exposure 
- Wait for auto exposure terminated
- Read auto-exposure value
- Stop record
- Config values manually with previous auto-exposure
- Record a bag file without auto-exposure 
- Stop record

Best Regards,

Sam,

MartyG-RealSense, 2019-04-09:
I hope the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

SamConti, 2019-04-10:
Thank you for the link.
Sincerely, I don't know C++, is there a way, an example...in python ?

MartyG-RealSense, 2019-04-10:
There is a Python script in this link that can be adapted to use the instructions for auto-exposure:

https://github.com/IntelRealSense/librealsense/issues/2021#issuecomment-406227519

SamConti, 2019-04-11:
OK thank you.
I understand what doing to read metadata, I tried this:

```
import, start, etc...
frameset = pipeline.wait_for_frames()
metadata = frameset.get_depth_frame().get_frame_metadata(rs.frame_metadata_value.actual_exposure)
```

but I have this error:
`RuntimeError: metadata not available`

So I try to recompile the library with the flag `ENFORCE_METADATA=true` but same error.

To recompile:

```
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DENFORCE_METADATA=true
make -j4
sudo make install
cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3) -DENFORCE METADATA=true
make -j4
sudo make install
```

I don't know if `DENFORCE_METADATA=true` have to be place for the realsense SDK installation (first cmake) or for the python bindings (second cmake) or the 2, so I placed it for each cmake.

ev-mp, 2019-04-11:
@SamConti  hello,
Reading metadata attributes, such as Depth actual exposure,  is an advanced capability provided by Realsense Depth cameras and (on Linux) requires kernel patches/Debian dkms package installed. Please review the [Metadata installation chapter](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#os-support)

Note that `frame.get_frame_metadata` will throw every time the requested attribute is not available. This is by design; you may find more  useful info in [Metadata documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md) provided.
In order to avoid raising exception we provide a complementary `supports_frame_metadata` function that shall be used as:
> if frame.supports_frame_metadata(attrib_x)
             frame.get_frame_metadata(attrib_x)

Also note that the metadata attributes are not guaranteed for each and every frame, as it is controlled by FW internal flows ( see the above link).

Regarding the build:
- Metadata functionality for Linux is always built-in and does not require additional CMake flags. But it does require kernel modifications (see above).
- The flag `ENFORCE_METADATA` is for Win builds only and has no effect on Linux, and is designed to notify the user in case the WinSDK version installed (required for metadata functionality) is not compliant with metadata requirements.

SamConti, 2019-04-15:
Hello,

Thank you for all explanations.

I see with` uname -r` I have `4.4.38-v7+` so my kernel is good...
Next I try to apply the patch with `./scripts/patch-realsense-ubuntu-lts.sh` but I have this error `gzip no space left on device`
I understand it's a space problem on boot, with `df` I have this line :
`/dev/mmcblk0p1   64456    32152    32304    50%   /boot`

So is there a way to apply this patch avoiding this problem ?

ev-mp, 2019-04-15:
@SamConti ,several things here:
The script you've tried to run is for official Ubuntu desktop LTS version with X86 arch, as specified in the [installation guide](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#linux-ubuntu-installation) and will not work in your environment.
You haven't specified the distro you're using, but apparently it is ARM-based distro with  `4.4.38-v7+`.
The system is also memory bounded, probably Rockchip RK3288/3229 or similar.

Note that ARM-based distros are not officially supported, but we'll try to assist to porting the patches if you decide to do so. You can also loop up this forum for related discussions.

Making kernel changes to such platforms may require rebuilding the kernels, as these are often build monolithic.  Additionally compiling on target is often not feasible due to memory/CPU constrains, so it also requires establishing cross-compiler.
Altogether it seem as a labor intensive task so it is worth reviewing whether following that path will benefit you.

SamConti, 2019-04-17:
Hello,

I wrote Raspberry Pi platform in the description and I think it was enough, sorry.

I understand you explain, so for the moment, as you say, it seems to be complicated and I don't follow this adaptation of the kernel.

ev-mp, 2019-04-18:
Thank you for the understanding
Feel free to reopen should you decide to port the patches.

closed 2019-04-18 by ev-mp