realsenseai/librealsense #3721: T265 pose diverges in standstill

Jonas-Witt, 2019-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5448 | 
| Operating System & Version | Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  build from latest source snapshot + patches                       | 
| Language                            |  C++                       | 
| Segment			|  Robot                      | 

### Issue Description
We are using the T265 on a robot to track its motion. We noticed that in standstill after a while, the velocity estimate diverges and the pose accelerates away into infinity. In the end the pose consists of NaN's. We are using the realsense ROS node to get the pose. I am currently running an experiment to see if this also happens in the realsense viewer.

bfulkers-i, 2019-04-10:
@Jonas-Witt are you using wheel odometry inputs to T265?

sirisian, 2019-04-10:
> I am currently running an experiment to see if this also happens in the realsense viewer.

It does. I assumed this was just an early bug. I was playing around with the sensor and it'll randomly shoot off into infinity for no reason. Half wondering if it's related to pointing it at LED lightbulbs. Would have to experiment more though.

Jonas-Witt, 2019-04-10:
We are currently doing first experiments with feeding wheel odometry in, but those observations were done without it. Anecdotally, I did not see the divergence when running the realsense viewer over night (last night). With the ROS node it usually happened within an hour (often within 10-20 minutes).

JozefRSTU, 2019-04-11:
I've had similar experiences, after some times it stops to track and pose gives me only NaN. After discussion with Intel support they told me to return device as it may be broken unit, but now I don't think so. When did you buy your T265? Mine was from very first batch ... I should receive two new devices tomorrow for tests to check if it is common issue.

Jonas-Witt, 2019-04-12:
Ours were also from the first batch. Hm, but if it's really a bad batch, what hardware change could lead to this?

JozefRSTU, 2019-04-12:
I've tested two new T265 (both with fw 0.0.18.5448) and both of them eventually stop tracking, but subjectively it is not so often as it was in my first device .... Realsense guys are aware of this issue, so hope they will solve it soon

Jonas-Witt, 2019-04-17:
I also saw a significant improvement. In earlier firmware versions, the divergence would happen quite quickly.

jsseng, 2019-04-23:
Sorry to ask a question in this thread, but I see you are using a TX2.  Did you have to patch the kernel to use kernel 4.9.140 with the T265 camera?  I cannot get the TX2 to detect the T265 camera.  Thanks.

JozefRSTU, 2019-04-23:
T265 doesn't need patched kernel, but you have to install udev rules for T265.

jsseng, 2019-04-23:
I have the file 99-realsense-libusb.rules in /etc/udev/rules.d, but in dmesg, I only see a Movidius device.  Do you show the Realsense device in dmesg?  I am plugged in the TX2 through a USB 3.0 hub, so I do not know if that is the issue.

JozefRSTU, 2019-04-23:
T265 doesn't work with all USB3.0 HUBs you should try another USB HUB or connect T265 directly into USB port on Jetson. Mine works well with TP-LINK UH700 USB HUB.

jsseng, 2019-04-23:
Thanks!  That is the tip I have been looking for.  I will try out that hub.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

Hi @Jonas-Witt

Have you had a chance to review with the latest librealsense SDK 2.21.0 version?  There has been updates to the FW specifically around re-localization.

Jonas-Witt, 2019-05-09:
Thanks for the update! While not having rigorously tested it recently, I have not seen the divergence in the last two weeks.

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Thanks for the feedback.

closed 2019-05-24 by RealSenseCustomerSupport

mickeyouyou, 2019-07-27:
I use librealsense SDK 2.21.0 version, and meet this problem this week.unstable pose data, 4 times 2 days.