realsenseai/librealsense #3722: [Legacy] V4L with R200 does not work

Maghoumi, 2019-04-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 | 
| Firmware Version                   | Latest | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra | 
| Platform                                 | Jetson Nano  |
| SDK Version                            |  legacy | 
| Language                            |  C | 
| Segment			|  Others | 

### Issue Description
I'm on Jetson Nano, but I think I gave this a try a few days ago on x86 running Ubuntu 16.04 with the same results.

I was able to compile librealsense and run (e.g.) `cpp-config-ui` without any issues. My apps all work fine if I use librealsense.  

Issue is that I can't seem to be able to use the RGB camera through any V4L application (e.g. `cheese`). What I do is:

``` 
cheese /dev/video0  # with or without sudo
```
I get: `Call to S_FMT failed for GREY @ 640x481: Input/output error`

It's been a while since I've used my R200, so I don't recall if I was ever able to use it through V4L. Any help is appreciated.

MartyG-RealSense, 2019-04-10:
The R200 is designed to be a rear facing (or "world facing") camera that faces away from the user.  A typical webcam that faces towards the user is a "front facing" camera, like the RealSense F200 and SR300.  In some non-Librealsense video-using applications such as Skype, the R200 can only be used as a 'rear camera' that provides an alternate video view that can be toggled to if a front-facing RGB camera is already active as the main video camera.

Maghoumi, 2019-04-10:
Thanks for your response. I'm not sure if my problem is strictly due to the "front facing vs. back facing camera" issue you mentioned. Unless I'm mistaken, my reasoning is that on the Jetson, dev apps that work with Raspberry Pi camera don't usually care about the distinction you mentioned.

I tried it with a C++ app that works with the Raspberry Pi camera through `gstreamer` and I still can use any of `/dev/video0` or `/dev/video1` or `/dev/video2`.

dorodnic, 2019-04-10:
Hi @Maghoumi 
Unlike more modern D400 / SR300 devices, the R200 is not a UVC compliant camera. 
To be more specific, R200 expects all sensors to be started / stopped together, and global configuration to be set (via a proprietary command) ahead of time (see [ds-device.cpp:316-354](https://github.com/IntelRealSense/librealsense/blob/legacy/src/ds-device.cpp#L316-L354)) 
You could probably extract the `stream_intent` command into a shorter script outside librealsense and run it prior to streaming.

Maghoumi, 2019-04-11:
@dorodnic Thanks, I thought this might be the case before I read your response. I tried running a librealsense app and at the same time open `cheese`, I saw a different error message which led me to believe the streams need to be open. I then read your response which confirmed my theory :)

For what I wanna do, I think it might be easier to just modify everything and use librealsense instead of V4L and gstreamer.

Thanks again, closing this issue

closed 2019-04-11 by Maghoumi