realsenseai/librealsense #3723: Real sense viewer scale does not show full movement

ShettyHarapanahalli, 2019-04-10:
| | |
|---|---|
|**librealsense**|2.19.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense T265|
|**Serial Number**|845412111034|
|**Firmware Version**|0.0.18.5129|

Hi, 

I am using the T265 for a AGV application in field, when i move take the t265 for long (300mtrs) and record the data, I am not able to visualize the data in 3d, can you please provide a means to view large scale data like 1 sq km or bigger. 

BR,
SSH

dorodnic, 2019-04-10:
Hi @ShettyHarapanahalli 
Please try `rs-trajectory` demo.

ShettyHarapanahalli, 2019-04-10:
Can we save the map or load the pre-recorded bag file with this.

dorodnic, 2019-04-10:
No, but you can always modify any of the SDK tools and examples to your needs. 
We will try to improve visualization in the Viewer, but it can take couple of releases to complete.

closed 2019-05-03 by RealSenseCustomerSupport