realsenseai/librealsense #3724: D435 depth breaking up at 60fps

Repulser, 2019-04-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Ubuntu 16.04, happened on MacOS aswell  | 
| Platform                                 |  NVIDIA Jetson TX2 |
| SDK Version                            |  2.18.0                           | 
| Language                            |  Python                         | 

### Issue Description
Hey, Im using the Realsense D435 alongside the Jetson TX2 and noticed a weird issue when running it at 60fps.
Im using 424x240 for the color stream and 480x270 for the depth stream.
On 30fps everything works perfectly but when changing to 60fps the depth stream breaks every few frames and returns 0 distance for any point in the frame.

Heres a link to the depth frame when running at 30fps: https://imgur.com/a/oKQB2Y2
And a link showing the depth frame at 60fps: https://imgur.com/a/Lo4ZS09

Would appreciate any insight, Thanks!

dorodnic, 2019-04-10:
Hi @Repulser 
Could you please try [firmware 5.11.4](https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family)? I'm pretty sure this was a firmware issue that got resolved, not 100% in what firmware version.

Repulser, 2019-04-11:
Updated and it works perfectly now, thanks and keep it up!

Repulser, 2019-04-16:
This issue seems to be happening again even on latest firmware (5.11.1.100)
It seems fine on the viewer but when using pyrealsense2 it fails

ev-mp, 2019-05-01:
@Repulser hello, 
Did you follow the instructions in #3786 to upgrade the FW to v5.11.6.200 ? 
What is the current status?

closed 2024-03-29 by MartyG-RealSense