realsenseai/librealsense #3728: "Failed to query (GET_CUR)" notices in syslog

mikepurvis, 2019-04-10:
I'm seeing a batch of these show up in my syslog when the camera initially connects (D435, on Ubuntu 16.04 with kernel 4.15). I get a burst of them when I connect with the Intel-supplied ROS driver, and also when running a standalone utility like `rs-enumerate-devices`. See: 

```
[ 2762.965743] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 2763.068518] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 2763.123279] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 2763.180513] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 2763.478541] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).
[ 2763.661804] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 2763.726968] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

They seem benign, so perhaps they just need to be downgraded to debug level, but either way it would be nice to understand what these are and get rid of them.

I looked briefly and didn't see an existing ticket for this, though there are some mentions of it in other discussions, eg https://github.com/IntelRealSense/librealsense/issues/3545#issuecomment-474773082, https://github.com/IntelRealSense/librealsense/issues/3374#issue-416797128.

SamLewis0602, 2019-04-15:
I'd just like to add that we also see these messages a lot with a D415 on Ubuntu 18.04.

RealSenseCustomerSupport, 2019-04-24:
----------------------------------------------

Hi mikepurvis, SamLewis0602,

Thanks for the feedback.
Wonder what FW versions you have with the cameras?
Also did you try patching up kernel and apply the udev rules (mentioned here: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)?

Thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi mikepurvis, SamLewis0602,

Do you guys have any update on this one?

If nothing else is needed, this one will be closed.

Thanks!

SamLewis0602, 2019-05-07:
We haven't had a chance to try the instructions at the above link, so you can close this ticket for now and I'll open a new one if we have issues. Thanks!

closed 2019-05-15 by RealSenseCustomerSupport

mikepurvis, 2019-10-15:
Still seeing this issue with firmware `05.11.06.250` and `05.11.15.00`.

smnogar, 2019-11-19:
I am also running into this running ubuntu 18. I see these messages while ROS is running. There is anecdotal evidence this is causing my systems to reboot. These messages were the last thing logged into the system log before my system rebooted. These events seem to be fairly rare though.

FrankFC, 2019-11-28:
Hello, I am getting the same error in the syslog and our host device enters in a reboot loop.
Our host device is:
SBC Up Squared - Intel Pentium N4200 
Intel Graphics 505
Ubuntu 16.04 LTS 64-bit.

I am using a D415 with fw 05.11.06.250.

Can you please re-open this ticket?

SoraDevin, 2020-02-11:
@RealSenseCustomerSupport can you reopen this please? I'm also seeing this issue.

sgstreet, 2020-02-16:
I'm seeing the same thing.  It appears to be related to the RS2_OPTION_GLOBAL_TIME_ENABLED attribute, I hundreds of of these message a minute.
`Feb 16 13:59:34 robot-dev kernel: [1209655.255174] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Feb 16 13:59:35 robot-dev kernel: [1209655.575976] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Feb 16 13:59:37 robot-dev kernel: [1209657.793412] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Feb 16 13:59:37 robot-dev kernel: [1209658.289817] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Feb 16 13:59:37 robot-dev kernel: [1209658.343556] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Feb 16 13:59:37 robot-dev kernel: [1209658.397033] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Feb 16 13:59:38 robot-dev kernel: [1209659.053319] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).`

lucbettaieb, 2020-10-16:
I also see this being spammed on `dmesg`.

doisyg, 2020-11-14:
Same here despite kernel patch on Ubuntu kernel 5.4. Why is this issue closed ? Is there a solution ?