realsenseai/librealsense #3730: librscalibrationapi for ubuntu bionic

realtime-neil, 2019-04-10:
```
$ cd $(mktemp -d) && for distribution in xenial bionic; do mkdir -vp "${PWD}/${distribution}"; curl -fsSLo "${PWD}/${distribution}/Packages" "http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/${distribution}/main/binary-amd64/Packages"; done && grep -nrEA1 '^Package: librscalibrationapi$' "${PWD}"
mkdir: created directory '/tmp/tmp.1bYZg0Y8cu/xenial'
mkdir: created directory '/tmp/tmp.1bYZg0Y8cu/bionic'
/tmp/tmp.1bYZg0Y8cu/xenial/Packages:5779:Package: librscalibrationapi
/tmp/tmp.1bYZg0Y8cu/xenial/Packages-5780-Version: 2.5.2.0
--
/tmp/tmp.1bYZg0Y8cu/xenial/Packages:5793:Package: librscalibrationapi
/tmp/tmp.1bYZg0Y8cu/xenial/Packages-5794-Version: 2.6.4.0
--
/tmp/tmp.1bYZg0Y8cu/xenial/Packages:5807:Package: librscalibrationapi
/tmp/tmp.1bYZg0Y8cu/xenial/Packages-5808-Version: 2.6.8.0
```

Looks like you have `librscalibration` for Ubuntu Xenial. Are there any plans to package it for Bionic? If so, when?

tim-rtr, 2019-04-11:
Just to provide some background for this issue, we are currently in process of upgrading to Ubuntu 18.04. Some of our packages have a dependency on `librscalibrationapi`. Obviously we can't build on 18.04 with this missing dependency. So this is unfortunately blocking our upgrade to 18.04.

tim-rtr, 2019-04-17:
Hi again. We would really appreciate a response. Is it very difficult to package `librscalibrationapi` for 18.04? Or is it just not prioritized? And if so, can we convince you to please give it more priority? :grin:

RealSenseCustomerSupport, 2019-04-23:
----------------------------------------------

Hi @tim-rtr,
The request has been forwarded to our engineering team. The priority and schedule will be given accordingly and we will keep you posted as soon as the schedule or any further update becomes available.
Thank you

eirikaso, 2019-05-08:
Following this issue. 
Please let us know if there are any updates to this issue, as I need to use this API.
If it is released soon it would be great :D

realtime-neil, 2019-05-08:
@eirikaso , if all you need are the libraries, then you can use the Xenial package. The shared objects require only standard symbols that are present on both Ubuntu versions:

```
$ cd $(mktemp -d) \
> && apt-get -y download librscalibrationapi \
> && find . -type f -name '*.deb' -print -exec basename -s .deb {} \; | xargs dpkg-deb --raw-extract \
> && find . -type f -name '*.so' -exec readelf -d {} + | grep -E '(^File|NEEDED)'
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librscalibrationapi amd64 2.6.8.0 [15.6 MB]
Fetched 15.6 MB in 1s (8525 kB/s)
File: ./librscalibrationapi_2.6.8.0_amd64/usr/lib/libDSDynamicCalibrationAPI.so
 0x0000000000000001 (NEEDED)             Shared library: [libdl.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libz.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
File: ./librscalibrationapi_2.6.8.0_amd64/usr/lib/librs2-crw-mm.so
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]

```

closed 2019-06-14 by RealSenseCustomerSupport