realsenseai/librealsense #3732: nodejs examples for T265 pose data. Help please!

eljefe40, 2019-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |T265| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux Ubuntu 18) | 
| Kernel Version (Linux Only)    |  (e.g. 4.16)                                         | 
| Platform                                 | NUC  |
| SDK Version                            |   legacy / 2.20.0                          | 
| Language                            |  nodejs                         | 
| Segment			| others                           |

I am unable to find working example for nodejs wrapper in order to capture pose data. The provided binary "rs-viewer" is working fine as to prove my connection is working. I am able to use python wrapper examples (not an option for this project) to get pose data but.. when I look for anything related to nodeJS, I only find examples that do not support T265. in fact the nodejs wrapper folder has had very little update in the past year or so. 
nodejs-capture.js in the examples folder (and referenced in the setup guides to test things out) will run ,but does not seem to be compatible with T265).
Im just looking for the most basic example, something like rs-pose or nodejs capture that actaully support T265 in order to observe properly formed api requests.
Any help would be appreciated. I have spend many many hours trying to "gleen' the appropriate API syntax from other wrapper but have fallen short of success.
kind regards

eljefe40, 2019-04-11:
I suppose its fair to say that nodeJS wrapper is not officially supported.

sirisian, 2019-04-17:
I just looked at the wrapper and pulled out the pieces, so I haven't tested the code below. It sounds like you're getting further than me if you can run nodejs-capture.js. I get a segmentation fault when calling context.create() which is like the first thing that's done in the wrapper.

I configured as it says before building, so I'm not sure what the issue is.
```cmake ../ -DBUILD_NODEJS_BINDINGS=1 -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true```

rs-pose and such works fine. Bit bewildering.

Maybe you can use the code below. Assuming I understand the logic correctly you create a context, then get the devices and each device has sensors that you get. From what I can tell the context, devices, and sensor objects are not GC as they reference C++ objects, so in real code you need to call destroy() on them all. Not a huge issue until you start using the context.setDevicesChangedCallback to detect unplugging and plugging in devices. I digress, see if you get further.
```js
'use strict';

const RS2 = require('bindings')('node_librealsense');

console.log('Creating context');
const context = new RS2.RSContext();
context.create();
console.log('Context created');

console.log('Query devices');
const devices = context.queryDevices();
console.log('Queried devices');

if (!devices)
{
	console.error('Error getting devices');
	process.exit(1);
}

// Loop through each device's sensors and look for a t265 sensor
let t265Sensor = null;
deviceLoop:
for (const device of devices)
{
	console.log('Query sensors');
	let sensors = device.querySensors();
	if (sensors)
	{
		console.log('Device has sensors');
		for (const sensor of sensors)
		{
			console.log('Found sensor');
			if (!sensor.isDepthSensor())
			{
				console.log('Found something that is probably a t265 sensor');
				t265Sensor = sensor;
				break deviceLoop;
			}
		}
	}
}

if (!t265Sensor)
{
	console.error('No t265 sensor connected');
	process.exit(1);
}

let poseFrame = new RS2.RSFrame();

t265Sensor.frameCallback = () =>
{
	if (poseFrame && poseFrame.isValid())
	{
		console.log('Handle pose frame');
		/*
		let profile = poseFrame.getStreamProfile();
		if (profile)
		{
			let streamValue = profile.stream();
			let formatValue = profile.format();
			let fpsValue = profile.fps();
			let indexValue = profile.index();
			let uidValue = profile.uniqueID();
			let isDefaultValue = profile.isDefault();
		}
		*/
		let pose =
		{
			translation: {x: 0, y: 0, z: 0},
			velocity: {x: 0, y: 0, z: 0},
			acceleration: {x: 0, y: 0, z: 0},
			rotation: {x: 0, y: 0, z: 0, w: 0},
			angularVelocity: {x: 0, y: 0, z: 0},
			angularAcceleration: {x: 0, y: 0, z: 0},
			trackerConfidence: 0,
			mapperConfidence: 0,
		};
		if (poseFrame.getPoseData(pose))
		{
			console.log(JSON.stringify(pose));
		}
	}
};
t265Sensor.startWithCallback('frameCallback', new RS2.RSFrame(), new RS2.RSFrame(), new RS2.RSFrame(), new RS2.RSFrame(), new RS2.RSFrame(), poseFrame);

t265Sensor.stop();
t265Sensor.close();
```

eljefe40, 2019-04-18:
Thanks for the reply. I was able to.. hold other language examples “up to the light” as well as studying  underlying dependency files and get my own version of res-pose working. It was very frustrating but ultimately gratifying to have it solved. I am not at my office today but can post my code if anyone cares to use it.

vnt1537, 2019-05-03:
I try to run t265_example.py from the python examples in wrappers and get an error for pipe.start(cfg) saying "Could'nt resolve requests". Is it not supported or do we have to make few changes?

closed 2019-05-03 by RealSenseCustomerSupport

ijeppsen, 2019-07-29:
eljefe40@ I'd love it if you could share your nodejs pose example. I'm starting down this path and banging my head against the wall.

SteinHDanCognite, 2019-09-21:
@eljefe40  I'd also love that code. I got a little bit further than the code above, but I seems to me there is something blocking access to the T265 further down. My D435 is not blocked, but my D435i also is (as with the T265). Anyways, here is what I have (in node_modules/node-librealsense):

```

const rs2 = require('./index.js');
const context = new rs2.Context();
const pipeline = new rs2.Pipeline();
const profile = pipeline.start();
const devices = context.queryDevices();
console.log('Found ' + devices.size + ' devices');
if (!devices) {
	console.error('Error getting devices');
	process.exit(1);
}

// Loop through each device's sensors and look for a t265 sensor
let t265Sensor = null;

for (var i = 0; i < devices.size; i++) {
	var device = devices.getDevice(i);
	console.log(device.getCameraInfo());
	console.log('Query sensors');
	let sensors = device.querySensors();
	if (sensors)
	{
		console.log('Device has sensors');
		for (const sensor of sensors)
		{
			console.log('Found sensor');
/*                        console.log(sensor.getCameraInfo());
			if (!sensor.isDepthSensor())
			{
				console.log('Found something that is probably a t265 sensor');
				t265Sensor = sensor;
				break deviceLoop;
			}
*/		}
	}
}

```

I also tried to add the device manually with the Config class:
```
var config = new rs2.Config();
config.enableAllStreams();
config.enableDevice('845112XXXXX'); // Serial number
```

But that also did not work

SteinHDanCognite, 2019-09-22:
I got it working with the latest SDK built from source and this code:

```
const rs2 = require('./index.js');

var config = new rs2.Config();
config.enableStream('pose', -1, 0, 0, '6dof', 0);
const pipeline = new rs2.Pipeline();
const profile = pipeline.start(config);

while (true) {
  var frameSet = pipeline.waitForFrames();
  frameSet.forEach(frame => {
    if (frame && frame.isValid && frame instanceof rs2.PoseFrame) {
      console.log(frame.poseData);
    }
  });
};

```

The device never shows in context.queryDevices(), but luckily it works anyways The output looks like this:

```
{ translation:
   { x: -0.0008122273720800877,
     y: -0.0030437964014708996,
     z: 0.0009263087413273752 },
  velocity:
   { x: 0.0032377545721828938,
     y: -0.023280125111341476,
     z: -0.00040748983155936003 },
  acceleration:
   { x: 0.032210610806941986,
     y: -0.19765973091125488,
     z: -0.05943886563181877 },
  rotation:
   { x: 0.20476378500461578,
     y: 0.666781485080719,
     z: -0.6782996654510498,
     w: 0.23104937374591827 },
  angularVelocity:
   { x: -0.01261275913566351,
     y: -0.0023317798040807247,
     z: -0.008244049735367298 },
  angularAcceleration:
   { x: -1.1702450513839722,
     y: -1.6991662979125977,
     z: 0.7158891558647156 },
  trackerConfidence: 2,
  mapperConfidence: 0 }

```