realsenseai/librealsense #3733: Multicam issue with Matlab

marcoagbarreto, 2019-04-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | {3.21.0.0} | 
| Operating System & Version |   {Win (10)}  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | {PC} |
| SDK Version                            |  { 2.19.1 }                          | 
| Language                            |  {Matlab}                          | 
| Segment			|  {others }                          | 

### Issue Description
<I'm using a multicam setup for a 3D capturing system with an array of 5 SR300 cameras using Matlab. I'm trying to acquire depth images by turning on 2 cameras at a time, while the others are turned off for the total of 5 cameras. The problem is that while using the realsense.pipeline(), the command automatically selects a realsense camera, and I can't adress to a specific camera. I tried communicating to each camera individually using Matlab with the multicam example https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam , but with no succes at all.  

Any help would be appreciated, thanks in advance. >

lramati, 2019-04-11:
Can you share your code?

marcoagbarreto, 2019-04-11:
> Can you share your code?

ctx = realsense.context(); % Create librealsense context for managing devices
n_devices = length(ctx.query_devices());
pipelines = realsense.pipeline();

for i = 1:n_devices
    
   pipe = realsense.pipeline(ctx);
   cfg = realsense.config();
   cfg.enable_device(dev.get_info(realsense.camera_info.serial_number))
   pipe.start(cfg);
   pipelines.get_active_profile(pipe)
    
end

lramati, 2019-04-11:
There appears to be some amount of missing code here. Where did dev come
from?

marcoagbarreto, 2019-04-11:
I was running this code to test the depth camera before trying to manage multicameras.  "dev" comes from this code.

    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    
    % define object for the point cloud
    pcl_obj = realsense.pointcloud();

    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();
    
    % Get streaming device's name
    dev = profile.get_device();
    name = dev.get_info(realsense.camera_info.name);

    % Obtener frames. Descartamos la primera pareja para permitir que
    % el tiempo de cámara para liquidar
    for  i = 1:5
        frames = pipe.wait_for_frames();
    end
    
   % Seleccione el marco de profundidad
    depth = frames.get_depth_frame();

    % obtener puntos de nube de punto sin color
    points = pcl_obj.calculate(depth);
    vertices = points.get_vertices();
    
    % Convertir los vertices en una nube de puntos
    ptCloud = pointCloud(vertices);

    % Guardar nube de puntos en un archivo PLY
    pcwrite(ptCloud,'3D_out.ply','PLYFormat','binary');

    % Mostrar nube de puntos
    pcshow(ptCloud);
    title(sprintf("Depth frame from %s", name));

    pipe.stop();

marcoagbarreto, 2019-04-11:
Hi, I just figure it out how to adress an specific camera, I went to the SDK and grab all of my cameras serial number and add them to a char array, later on I assigned the serial number to the config file, then made a loop to check the stream in all cameras. Here is the code to loop through all cameras just to check that they are connected.  Issue solved. Thanks

ctx = realsense.context(); % Create librealsense context for managing devices
n_devices = length(ctx.query_devices()); % Number of devices connected
cam_sn = ['627204001255';'627206001644']; % Serial numbers of all cameras

for i = 1:n_devices % Loop through all devices detected
    
   pipe = realsense.pipeline(ctx); % Make Pipeline object to manage streaming
   cfg = realsense.config(); % Create a new configuration
   cfg.enable_device(cam_sn(i,:)); % Enable device in config file with the serial number of the camera
   pipe.start(cfg); %Start streaming on a selected camera with default settings
   pipe.stop(); % Stop streamming
    
end

closed 2019-04-11 by marcoagbarreto