realsenseai/librealsense #3734: T265 can only be accessed by the first pipe instantiated

stephen-systemfriend, 2019-04-11:
----------------------------------------------------------------------------------------------------

|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            | 2.20                                                        | 
| Language                            |  C }                          | 

If you create multiple rs::pipe objects, only the first one can find the T265. I think that they must be locking it in the constructor somewhere. 

heres a MWE

`
#include <iostream>
#include <librealsense2/rs.hpp>  

int main()
{
	std::string T265_serial = "845412110352";
	std::string D435i_serial = "838212070412";
        //if you swap these two, it will no longer run
	rs2::pipeline T265_pipe;
	rs2::pipeline D425i_pipe;
	rs2::config T265_cfg, D435i_cfg;

	D435i_cfg.enable_device(D435i_serial);
	T265_cfg.enable_device(T265_serial);
	// Add pose stream
	T265_cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

	try
	{
		T265_pipe.start(T265_cfg);
		D425i_pipe.start(D435i_cfg);
	}
	catch (std::exception e)
	{
		std::cout << "failed to start pipes with error: " << e.what() << std::endl;
		return -1;
	}
	
	//diplay the stream
	for (int i = 0; i < 200; i++) {
		//pose
		auto pose_frames = T265_pipe.wait_for_frames();
		auto f = pose_frames.first_or_default(RS2_STREAM_POSE);
		auto pose_data = f.as<rs2::pose_frame>().get_pose_data();
		std::cout << "Device Position: "  << std::fixed << pose_data.translation.x << " "
			<< pose_data.translation.y << " " << pose_data.translation.z << " (meters)" << std::endl;
		//depth
		auto depth_frames = D425i_pipe.wait_for_frames();
		auto df = depth_frames.get_depth_frame();
		std::cout << "Depth Disparity: " << df.get_distance(10, 100) << std::endl;

	}
	return 0;
}
`

ev-mp, 2019-04-11:
@stephen-systemfriend hello, 
Currently all the connected T265 sensors are registered on startup with the first rs2::context created.
Hence, in order to to open multiple pipes that rs2::context should be shared among these pipes. It is achieved by passing the rs2::context to each pipe created.
> rs2::context                ctx;
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
...

Refer to [multicam](https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L15-L29) example for the full code.

Links to #3434, #3437

ev-mp, 2019-04-18:
@stephen-systemfriend , do you need more assistance with this ?

stephen-systemfriend, 2019-04-18:
the fix was perfect, thank you

closed 2019-04-18 by stephen-systemfriend

karry3775, 2020-01-14:
Were you able to launch multiple t265 with the help of ros. I am currently using realsense-ros package to launch one t265 (demo_t265.launch). I tried using the (rs_multiple_devices.launch) file, but it doesn't support two t265s at the time. Was there a fix you could find. Thank you

stephen-systemfriend, 2020-01-15:
@karry3775  I only had one t265 plugged in, so what you have is a different issue that you should probably open separately

karry3775, 2020-01-15:
Thank you for your response. I will open a new issue in that case.