realsenseai/librealsense #3735: How to get Aligned-to-Color depth Frame's  intrinsic parameters with Matlab wrapper?

kazu0622, 2019-04-11:
| Camera Model                       | {D400 } | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   {Win 10)  | 
| Platform                                 | PC|
| SDK Version                            | 2.20.0  | 
| Language                            | Matlab      | 
| Segment			|  others           | 

Hello.
I'd like to have some suggestions  how to get depth frame's intrinsic parameters ,that is alinged to Color frame with matlab wrapper.

I've already checked this issue below on python.
https://github.com/IntelRealSense/librealsense/issues/3688

Here is my script. I guess there're some mistake.
cfg = realsense.config();
cfg.enable_stream(realsense.stream.depth,1280,720,realsense.format.z16);
cfg.enable_stream(realsense.stream.color,1280,720,realsense.format.rgb8);
pipe = realsense.pipeline(); 
profile = pipe.start(cfg);
align_to_color = realsense.stream.color;
depth_stream = profile.get_stream(realsense.stream.depth);
depth_stream = depth_stream.as('video_stream_profile');
intrinsics1 = depth_stream.get_intrinsics();
% I guess this might be the original intrinsics before aligned.%

fs = pipe.wait_for_frames();
align = realsense.align(align_to_color);
aligned_frames = align.process(fs);
depth_tmp = aligned_frames.get_depth_frame();

Thanks.

ev-mp, 2019-04-14:
Hello @kazu0622 ,
You can use the intrinsic of the stream to which the depth is aligned to - e.g. color in the above snippet:

>color_stream = profile.get_stream(realsense.stream.color);
color_video_stream = rgb_stream .as('video_stream_profile');
color_intrinsic=depth_aligned_to_color_intrinsic = color_video_stream.get_intrinsic()

Note that you should also be able to get intrinsic directly from the "aligned_depth" frame in a similar manner (frame->frame_profile->video_frame_profile->get_intrinsic), but I don't have the Matlab environment to provide a reference code.

kazu0622, 2019-04-15:
@ev-mp Thank you for your quick response and your information about my issue.  I'll try it.

closed 2019-04-15 by kazu0622