realsenseai/librealsense #3738: Broken timestamps

elevir, 2019-04-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.19.2 / 2.20.0                          | 
| Language                            |  C++                          | 

Hello!

After patch 2.19.2 method frame::get_timestamp() returns broken timestamps.
There is code of obtaining frames:
```
void RealSenseDevice::RSCallback(const rs2::frame& frame)
{
	if (rs2::frameset frameset = frame.as<rs2::frameset>())
	{
		std::lock_guard<std::mutex> lock(_visualMutex);
		rs2::video_frame left = frameset.get_fisheye_frame(LEFT);
		rs2::video_frame right = frameset.get_fisheye_frame(RIGHT);
		_stroPairWasUpdated = left && right;
		if (_stroPairWasUpdated)
		{
			_curStereoPair.leftFrame = RealSenseUtils::frame_to_mat(left);
			_curStereoPair.leftFrame.timestamp = left.get_timestamp();
			_curStereoPair.timestamp = left.get_timestamp();
			_curStereoPair.rightFrame = RealSenseUtils::frame_to_mat(right);
			_curStereoPair.rightFrame.timestamp = right.get_timestamp();
		}
	}
	else
	{
		rs2::motion_frame motion = frame.as<rs2::motion_frame>();
		auto motionProfile = motion.get_profile();
		
		std::lock_guard<std::mutex> lock(_inertialMutex);
		if (motionProfile.stream_type() == RS2_STREAM_ACCEL && motionProfile.format() == RS2_FORMAT_MOTION_XYZ32F)
		{
			auto accelerationVec = motion.get_motion_data();
			std::memcpy(&_curPose.vecAcc.a[0], &accelerationVec, sizeof(rs2_vector));
			_curPose.timestamp = motion.get_timestamp();
			_poseWasUpdated = true;
		}
		if (motionProfile.stream_type() == RS2_STREAM_GYRO && motionProfile.format() == RS2_FORMAT_MOTION_XYZ32F)
		{
			auto gyroVec = motion.get_motion_data();
			std::memcpy(&_curPose.vecVelocity.a[0], &gyroVec, sizeof(rs2_vector));
			_curPose.timestamp = motion.get_timestamp();
			_poseWasUpdated = true;
		}
	}
}
```
The timestamp for both visual and inertial data has value -2147483648 always. Note that, the data is correct, but timestamps are broken.
With patch 2.19.1 all works fine.
I suppose that this problem appeared after PR #3453. Do I need to install any T265 drivers for enabling system timestamps or is it bug?

Thanks!

DaddyWesker, 2019-04-22:
Hey. Have the same issue. WIn 10 x64

ev-mp, 2019-04-24:
@elevir hi, 
The code fragment seem to be legit, and yet we couldn't find a way to reproduce it so far.
Do you see it with the standard `realsense-viewer` tool? 
If so - in which versions 2.19/2.20/2.21 ?

[image]

I'd suggest to start by putting `cout` s next to query to see that the data is not altered elsewhere in the code:
```c++
	rs2::video_frame left = frameset.get_fisheye_frame(LEFT);
	rs2::video_frame right = frameset.get_fisheye_frame(RIGHT);
	if (left) std::cout << left.get_timestamp();
	if (right) std::cout << right.get_timestamp();
```

elevir, 2019-04-24:
@ev-mp, hello!

I took screenshots of viewers since version 2.19.1. For versions above 2.19.1 timestamps have warning that the hardware timestamps are unavailable.

> I'd suggest to start by putting cout s next to query to see that the data is not altered elsewhere in the code:

I've already checked, they were broken.

2.19.1 - 
[image]

2.19.2 - 
[image]

2.20.0 - [image]

2.21.0 - 
[image]

ev-mp, 2019-04-24:
@elevir  hi,
At least int the provide 2.19.2 screenshot it is clear that the IMU and Fisheye timestamps are not identical:
[image]

In other screenshot this in either trimmed or obscured so I couldn't confirm it, but it seem that the system timestamp are within the expected range.
Can you provide some 5-10 sec record file of the streaming?
[image]

ev-mp, 2019-04-24:
Note also that the value that you've mentioned in the beginning  - 
>-2147483648

which is obviously 0x80000000, is definitely different from the timestamps captured in the screenshots

elevir, 2019-04-24:
@ev-mp 

> Can you provide some 5-10 sec record file of the streaming?

Unfortunately, I will able to provide recorded file just tomorrow.

> Note also that the value that you've mentioned in the beginning -
> 
> >-2147483648
> 
> which is obviously 0x80000000, is definitely different from the timestamps captured in the screenshots

That's true, but I don't know how is realized timestamp capturer in RealSense viewer. Can you confirm that there is no any substitution of hardware timestamps by software timestamps?

elevir, 2019-04-25:
@ev-mp hello!
The recorded file is [here](https://yadi.sk/d/suhXnaz3hhKa0A)

ev-mp, 2019-04-29:
@elevir, I checked the record and the timestamps for Fisheye and IMU streams are fully functional:
[image]

Both timestamps and frame numbers update as expected.
There is a problem I encounter with the playback functionality that does not show pose frame metadata (frame number, timestamp, fps) - but that's a separate issue.

Please double check the code you use - e.g. things like frame-frameset casting.
The Viewer's code in this regards is written in a [straight-forward manner](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L1690) though as overall not that easy to follow:
>frame_number = f.get_frame_number();
        timestamp_domain = f.get_frame_timestamp_domain();
        timestamp = f.get_timestamp();

Therefore I'd recommend to check `rs-data-collect` tool with T265 , inspect the timestamps and [review the tool's code](https://github.com/IntelRealSense/librealsense/blob/master/tools/data-collect/rs-data-collect.cpp#L144-L149). That code base is much smaller and provides similar functionality.

elevir, 2019-04-29:
@ev-mp ok, thanks!
I'll try to implement in the same way as in this tool, and get back to you as soon as possible.
But anyway in framework with version 2.19.1 the code I showed was working.

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

@elevir
Have you been able to verify this?  Can the issue be closed?

elevir, 2019-05-06:
@RealSenseCustomerSupport I'm sorry, I didn't verify this yet. I will try to do this tomorrow.

elevir, 2019-05-07:
@ev-mp hello!
I've executed the code below with different versions of SDK:
```
void omw::device::RealSenseDevice::RSCallback(const rs2::frame& frame)
{
	if (rs2::frameset frameset = frame.as<rs2::frameset>())
	{
		std::lock_guard<std::mutex> lock(_visualMutex);
		rs2::video_frame left = frameset.get_fisheye_frame(LEFT);
		rs2::video_frame right = frameset.get_fisheye_frame(RIGHT);
		_stroPairWasUpdated = left && right;
		if (_stroPairWasUpdated)
		{
			_curStereoPair.leftFrame = RealSenseUtils::frame_to_mat(left);
			_curStereoPair.leftFrame.timestamp = left.get_timestamp();
			std::cout << "left timestamp: " << _curStereoPair.leftFrame.timestamp << std::endl;
			_curStereoPair.timestamp = left.get_timestamp();
			_curStereoPair.rightFrame = RealSenseUtils::frame_to_mat(right);
			_curStereoPair.rightFrame.timestamp = right.get_timestamp();
			std::cout << "right timestamp: " << _curStereoPair.rightFrame.timestamp << std::endl;
		}
	}
	else
	{
....

....
        std::cout << "Real sense initializing..." << std::endl;
	rs2::config cfg;
	cfg.disable_all_streams();
	cfg.enable_stream(RS2_STREAM_FISHEYE, LEFT, IMG_IN_WIDTH, IMG_IN_HEIGHT);
	cfg.enable_stream(RS2_STREAM_FISHEYE, RIGHT, IMG_IN_WIDTH, IMG_IN_HEIGHT);
	cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
	cfg.enable_stream(RS2_STREAM_GYRO,  RS2_FORMAT_MOTION_XYZ32F);
	rs2::pipeline_profile profile = _pipe.start(cfg, std::bind(&RealSenseDevice::RSCallback, this, std::placeholders::_1));
....
```
Output with patch 2.19.1:
[image]

Well, as we can see timestamps are correct.

Output with last patch (2.21.0)
[image]

Also, I've noticed, if I build library with parameter `-DBUILD_WITH_TM2=OFF` then camera is not detected, but if I leave this parameter with default value (I suppose that default value is ON) then camera is detected successfully, but timestamps are incorrect. Is this a coincidence?

elevir, 2019-05-07:
@ev-mp  Oh, my god. Timestamps are double :D

closed 2019-05-07 by elevir