realsenseai/librealsense #3740: Getting Depthvalue in Disparitymode

Gnaarki, 2019-04-11:
In my project I use a D435 and I need the disparitymode, but I need the depthvalues too, is it possible to get the depthvalues from the disparityvalues?

I have the disparity as a cv::Mat type.

the only things I can find in librealsense is
(_stereo_baseline_meter * _focal_length_mm * fractions) / _depth_units)
but I don't know how i should use this in the transformation and I don't know what fractions is

I found this too, but it gives wrong depthvalues depth[disparity > 0] = (fx * baseline) / (units * disparity[disparity > 0])

Gnaarki, 2019-04-14:
As an example:

If I have a baseline of 0.049778m, a focal length of ~617, fractions is hard-coded with 32 and depth_units is 0.001. Then I get 982817 as factor. The disparity is max 65536, when I divide the factor by 65536 I get a minZ of ~15m

This example is based on the disparity_transform::update_transformation_profile method in disparity-transform.cpp

_d2d_convert_factor = (_stereo_baseline_meter * _focal_lenght_mm * fractions) / _depth_units

ev-mp, 2019-04-14:
@Gnaarki hello,
In general, to obtain disparity data in Librealsense you either had to start with a depth frame and then convert it to disparity domain, or use a 3rd party code.
Hence please elaborate how you obtained the disparity data:
>I have the disparity as a cv::Mat type.

Did you get a depth frame an run it though the disparity_transform block ?  
In that case you can follow the implementation and convert depth<->disparity values.
You can also have Depth and Disparity frames side by side and use them as needed:
```c++
rs2::disparity_transform depth2disparity;
auto frameset = pipe.wait_for_frames();
if (auto depth_frame = frameset.get_depth_frame())
{
    auto disparity_frame = depth2disparity.process(depth_frame );
   ...
   Write your code here
   ....
}
```

* Advanced stuff - there is also a developers' option (not officially supported) to get the raw disparity data in 1/32 pixel units directly from the device (a.k.a. "**scaled disparity**"). Note that invoking this option alters the content of the depth frame, therefore you must be extra-cautious of how to interpret the data. See #3344 for details.

ev-mp, 2019-04-18:
@Gnaarki hi, do you need more assistance with this ?

Gnaarki, 2019-04-18:
@ev-mp 
I changed the order, I use the depthimage and convert it to disparityimage, now it works fine.

closed 2019-04-18 by Gnaarki