realsenseai/librealsense #3741: The depth quality get worse in the edge of object

TouchDeeper, 2019-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I want to do a pose estimation based on point cloud. My object is a banana.
The scene I get the point cloud of banana is shown in the image below:
[image]

The point cloud I get from the point cloud  is shown in the image below:
view along the Z axes:
[image]

side view:
[image]

As you can see from the side view,  the point cloud in the edge will become elongation along the Z axes.
What can I do to preserve the point cloud in the edge of banana?

TouchDeeper, 2019-04-13:
I have applied the spatial filter and the temporal filter.
The preset I use is the [ShortRangePreset.json](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets)

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

Hi TouchDeeper,

I think that you are on right track. And below are some general things that I can suggest:
1) If not yet, make sure to validate your camera's depth quality by going through Depth Quality Tool. And then do calibration if needed.
2) Starting with a baseline (either the default preset as it is tuned for best trade-offs among all parameters, or use the ShortRangePreset). And add no enabling post-processing;
    Also use 1280x720 fpr D415, and 848x480 for D435.
You may compare with a few different presets (especially for high density if you are looking for edge preserving);
3) Keep nice lighting environment and laser on. Increase laser power if needed.
4) Turn on post-processing with spacial/temporal filter on, and also try other post-processing filters.
5) You may also adjust depth control parameters under "Advanced Controls => Depth Control"

And refer to this whitepaper about tuning depth quality, if not yet: https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

Again, edge preserving is a tough topic and there are various factors to consider.
Hope this can be helpful.

Thanks!

TouchDeeper, 2019-04-23:
Thanks, we also have a discussion in the [Intel forum](https://forums.intel.com/s/question/0D70P000006H82gSAC), and when I get closer from object to the camera, the depth quality of the edge of the object become better.

closed 2019-04-23 by TouchDeeper

tugbakara, 2023-07-19:
In this [link](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) , If I set this parameters in the viewer will they help in ROS side?

tugbakara, 2023-07-19:
Really thanks! I will try an give a fedback! @MartyG-RealSense  :)