realsenseai/librealsense #3742: Installation error on Linux（Ubuntu）

momo1986, 2019-04-12:
`sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE`
OK.

16-LTS and 18-LTS corresponding step： **Failed**
Log：

> Get:1 file:/var/cuda-repo-9-0-176-local-patch-4  InRelease
> Ign:1 file:/var/cuda-repo-9-0-176-local-patch-4  InRelease
> Get:2 file:/var/cuda-repo-9-0-local-cublas-performance-update-2  InRelease
> Ign:2 file:/var/cuda-repo-9-0-local-cublas-performance-update-2  InRelease
> Get:3 file:/var/cuda-repo-9-0-local-cublas-performance-update-3  InRelease
> Ign:3 file:/var/cuda-repo-9-0-local-cublas-performance-update-3  InRelease
> Get:4 file:/var/cuda-repo-9-0-local-cublas-performance-update  InRelease
> Ign:4 file:/var/cuda-repo-9-0-local-cublas-performance-update  InRelease
> Get:5 file:/var/cuda-repo-9-0-local  InRelease
> Ign:5 file:/var/cuda-repo-9-0-local  InRelease
> Get:6 file:/var/cuda-repo-9-0-176-local-patch-4  Release [574 B]
> Get:7 file:/var/cuda-repo-9-0-local-cublas-performance-update-2  Release [574 B]
> Get:8 file:/var/cuda-repo-9-0-local-cublas-performance-update-3  Release [574 B]
> Get:9 file:/var/cuda-repo-9-0-local-cublas-performance-update  Release [574 B]
> Get:10 file:/var/cuda-repo-9-0-local  Release [574 B]
> Get:6 file:/var/cuda-repo-9-0-176-local-patch-4  Release [574 B]
> Hit:11 http://mirrors.aliyun.com/ubuntu bionic InRelease
> Get:7 file:/var/cuda-repo-9-0-local-cublas-performance-update-2  Release [574 B]
> Get:8 file:/var/cuda-repo-9-0-local-cublas-performance-update-3  Release [574 B]
> Get:9 file:/var/cuda-repo-9-0-local-cublas-performance-update  Release [574 B]
> Get:10 file:/var/cuda-repo-9-0-local  Release [574 B]
> Hit:12 http://mirrors.aliyun.com/ubuntu bionic-security InRelease
> Hit:13 http://mirrors.aliyun.com/ubuntu bionic-updates InRelease
> Err:20 http://realsense-alm-public.s3.amazonaws.com/apt-repo xenial InRelease
>   403  Forbidden [IP: 52.218.49.105 80]
> Hit:21 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
> Hit:22 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease
> Get:23 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease [2381 B]
> Get:24 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages [29.7 kB]
> Ign:15 https://developer.download.nvidia.cn/compute/machine-learning/repos/ubuntu1804/x86_64  InRelease
> Hit:25 https://developer.download.nvidia.cn/compute/machine-learning/repos/ubuntu1804/x86_64  Release
> Reading package lists... Done
> E: Failed to fetch http://realsense-alm-public.s3.amazonaws.com/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.49.105 80]
> E: The repository 'http://realsense-alm-public.s3.amazonaws.com/apt-repo xenial InRelease' is not signed.
> N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> N: See apt-secure(8) manpage for repository creation and user configuration details.

ev-mp, 2019-04-14:
Hello @momo1986 ,
Apparently the Librealsense repository you're using is not up to date:
>Err:20 http://**~~realsense-alm-public~~**.s3.amazonaws.com/apt-repo xenial InRelease
403 Forbidden [IP: 52.218.49.105 80]

Please follow the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) - should be  **realsense-hw-public**, and update.

durka, 2019-04-15:
http://realsense-hw-public.s3.amazonaws.com/apt-repo also gives a 403 Forbidden response. Is there a new URL? This issue seems to be closed and reopened every couple of weeks with no official response :(

ev-mp, 2019-04-16:
@durka hello, 
No, there are no planned changes for the repo URL. I also checked both Bionic and Xenial and both repos are fully functional.
Can you check that you're not behind corporate firewall that require proxy ? https://github.com/IntelRealSense/librealsense/issues/1670#issuecomment-389467743

@momo1986 , I wonder if you've managed to resolve the issue, please update.

durka, 2019-04-16:
> ? #1670 (comment)

No, I'm not behind any firewall. I tried the URL on three different
connections, including mobile data, and it's always 403. Have you tested
from an external IP?

ev-mp, 2019-04-16:
@durka , please post the full log when you try to access the repo:
>sudo apt-get update
>sudo apt-get install librealsense2-utils

durka, 2019-04-16:
It works after clearing out a bunch of entries from my sources.list. It seems if any repo fails, apt does not update anything.

closed 2019-04-22 by ev-mp