realsenseai/librealsense #3744: illumination problem

Mikeraka, 2019-04-12:
Hello, I have got some depth and rgb image by D435 as follows
[image]
[image]

the depth in the red circle in depth image above is obviously wrong, is it normal? is it because that the D435 is sensitive to illumination?

MartyG-RealSense, 2019-04-12:
The circled area corresponds to a strong reflection on the desk surface from the room lighting.  Reflections can make it harder for the camera to see the detail of a surface in the area where the reflection is occurring.

Mikeraka, 2019-04-13:
@MartyG-RealSense ok,I get it. Thank you very much for your answering.

closed 2019-04-13 by Mikeraka