realsenseai/librealsense #3746: Latest T265 Firmware?

FriskyBishop, 2019-04-12:
what is the latest T265 Firmware? 
My device has 0.0.18.5129 
I tried searching downloadcenter.intel.com but searching "t265" did not return any results

bfulkers-i, 2019-04-12:
Hi @FriskyBishop, please update to librealsense `2.20.0` which comes with FW version `0.0.18.5448`. For T265, the firmware is bundled with librealsense so to have the latest version you should always use the latest version of librealsense.

FriskyBishop, 2019-04-15:
@bfulkers-i where is the firmware located? 
I installed the SDK using the Intel.RealSense.Viewer.exe but I cannot locate the fireware files.

bfulkers-i, 2019-04-15:
@FriskyBishop, the T265 firmware is updated automatically by each librealsense version to match that version of librealsense, so as long as you use the newest SDK (currently `2.20.0`) and unplug and replug the T265, you are sure to be running the latest version.

FriskyBishop, 2019-04-15:
@bfulkers-i oh yeah, so it is xD
thanks, problem solved

closed 2019-04-27 by RealSenseCustomerSupport