realsenseai/librealsense #3751: RealSense Viewer LibGL issue on Jetson TX2

JLenssen, 2019-04-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  v2.9.1                         | 
| Language                            |  C                          | 
| Segment			|  Camera                          | 

### Issue Description
After compiling the librealsense  library, the RealSense Viewer cannot be started and is exiting with
"Could not initialize offscreen context!". This happens in [ux-window](https://github.com/IntelRealSense/librealsense/blob/master/common/ux-window.cpp). We suspect, that this is due to the LibGL version being to low. We found, that GLFW_VERSION_CONTEXT is set to 2.0 somewhere in the viewer code. When checking out v2.13.0 of the SDK, we can start the RealSense Viewer.  In the installation guide it is recommended to use `sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`, this seems to install the wrong libgl1 version. Camera is working and examples which do not use libgl were tested and are working.

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi JLenssen

Just to be clarified, your issue happens with RS 2.9.1 but not RS 2.13.0?

Thanks!

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Hi JLenssen,

Wonder if there is any update? Or do you still need help?

Thanks!

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi JLenssen,

If nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-05-24 by RealSenseCustomerSupport