realsenseai/librealsense #3752: Aligned Point Cloud possibly still utilizing wrong intrinsics

joki123, 2019-04-12:
I'm opening a new issue because it issue #2002 was closed out. I'm encountering the same issues utilizing the following code:

  rs2::pipeline_profile profile = pipe.start();
  float depth_scale = get_depth_scale(profile.get_device());
  rs2_stream align_to = find_stream_to_align(profile.get_streams());
  rs2::align align(align_to);

  for (auto i = 0; i < 30; ++i) {
    pipe.wait_for_frames();
  }

  auto processed = align.process(pipe.wait_for_frames());
  rs2::video_frame rgb_vf = processed.first(align_to);
  rs2::depth_frame raw_df = processed.get_depth_frame();

  rs2::pointcloud pc;
  rs2::points points;
  pc.map_to(rgb_vf);
  points = pc.calculate(raw_df);
  points.export_to_ply("test.ply", rgb_vf);

I received some assistance from Sergey Dorodnicov and we established that not aligned pointclouds can be generated fine and that the resulting pointclouds from above definitely appear to be misaligned.

Can anyone confirm that issue #2002 was indeed resolved? If so, is it possible the alignment code might not be working properly? Note: alignment code above is taken from the public examples.

joki123, 2019-04-13:
Confirming that I can generate a valid point cloud from aligned data so it appears the point cloud generation code is not utilizing the right intrinsic information.

ev-mp, 2019-04-14:
@Joki123 hello and thank you for the report and the additional info.
We're looking into it and it seem indeed that there is a potential extrinsic issue with the aligned stream.
We'll keep you updated.

ev-mp, 2019-04-15:
@joki123 , there is a fix PR being evaluated that should address the offset in pointcloud mapping from aligned depth.
Visually, the changes are subtle so I suppose it will require a controlled environment to measure the improvement
You can try it from [here](https://github.com/IntelRealSense/librealsense/pull/3760)

ev-mp, 2019-05-01:
@joki123 , the fix was merged into the recently released v2.21.0. Do you still observe discrepancies with that version ? Please update

joki123, 2019-05-01:
Thanks for the fix! It's greatly appreciated. I'll check it out. I've been
creating point clouds via a third-party library instead.

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

Hi @joki123,
Do you have a chance to check out v2.21.0?

joki123, 2019-05-15:
Actually the fix came a bit too late we ended up creating point clouds from
a third-party library. I'm sure this fixed will greatly help other people
however!

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi joki123,

Thanks for the updates. Glad to hear that you have already addressed the issue.
Closing this one.

Thanks!

closed 2019-05-22 by ev-mp