realsenseai/librealsense #3753: I need to get this working and I have an error, "hwmon command 0x10 failed. Error type: Invalid Command (-1)." / Raspberry Pi Model 3 B+ / D415 / Raspbian Stretch (Stretch 9.8 / Kernel 4.14.98-v7+)

NMovere, 2019-04-14:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400, (specifically D415) | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Raspbian Stretch Release 9.8 | 
| Kernel Version (Linux Only)    |  4.14.98-v7+ | 
| Platform                                 | Raspberry Pi Model 3 B+|
| SDK Version                            | 2.20.0, is what shows at the bottom of the RealSense Viewer| 
| Language                            | Python 3.5.3| 
| Segment			|              | 

### Issue Description
Hi, I followed this tutorial: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md to install the RealSense SDK on my Raspberry Pi Model 3 B+ with Raspbian Stretch (release 9.8; kernel 4.14.98-v7+) so that I can use the D415 camera with the Python wrapper of the SDK. The install seemed like it went successfully, I was able to import pyrealsense2 into my Python3 program. When I tried to run this following Python program:
`import pyrealsense2 as rs # worked perfectly, no errors after this statement.
pipeline = rs.pipeline() # worked perfectly, no errors.
pipeline.start() # Error occured here, this is the traceback: Traceback (most recent call last): File "<stdin>", line 1, in <module> RuntimeError: hwmon command 0x10 failed. Error type: Invalid Command (-1).`

When I run "realsense-viewer" in terminal, I get a similar error immediately popping up: 
`Invalid Value in rs2_create_device(info_list:0xdc05b0, index:0): hwmon command 0x10 failed. Error type: Invalid Command (-1).`

If I unplug / re-plug in the camera with the realsense-viewer open, along with the above error, another error shows under that one:
`(rs.cpp:279) null pointer passed for argument "device"`

- When running "lsusb" it shows "Intel Corp."
- The camera is plugged directly into the Raspberry Pi Model 3 B+, no externally powered USB attachment.

Please help, I need to get this working. Thank you.

ev-mp, 2019-04-14:
@NMovere  hello,
The error you receive indicates communication error with the basic device functionalities.
To rule out the possible HW setup issues, I recommend to connect and try this device on a different PC - in case the error can be isolated to Raspberry Pi then I'd try to switch to a self-powered USB hub and/or replace USB cable.

In addition I'd ask you to collect FW logs with `rs-fw-logger` tool:
- Run the tool
- Run `rs-enumerate-devices` or `rs-depth` in another terminal
- Collect and post the log obtained with `rs-fw-logger` .

NMovere, 2019-04-14:
@ev-mp 

Running **rs-fw-logger** also shows:

> **pi@raspberrypi:~ $ rs-fw-logger**
> 
> **Waiting for RealSense device to connect...**
> **RealSense error calling rs2_device_hub_wait_for_device(hub:0x1974368):**
>     **hwmon command 0x10 failed. Error type: Invalid Command (-1).**
> 
> **Waiting for RealSense device to connect...**
> **RealSense error calling rs2_device_hub_wait_for_device(hub:0x1974368):**
>     **hwmon command 0x10 failed. Error type: Invalid Command (-1).**
> 
> **Waiting for RealSense device to connect...**
> **RealSense error calling rs2_device_hub_wait_for_device(hub:0x1974368):**
>     **hwmon command 0x10 failed. Error type: Invalid Command (-1).**
> 
> **Waiting for RealSense device to connect...**

Running **rs-enumerate-devices** in another terminal shows:

> **pi@raspberrypi:~ $ rs-enumerate-devices**
> **RealSense error calling rs2_create_device(info_list:0x16c6020, index:0):**
>     **hwmon command 0x10 failed. Error type: Invalid Command (-1).**

Running **rs-enumerate-devices** also shows:

> **pi@raspberrypi:~ $ rs-enumerate-devices**
> **RealSense error calling rs2_create_device(info_list:0x8cb700, index:0):**
>     **Could not open device. Last Error: Device or resource busy**

And, running **rs-depth** shows:

> **pi@raspberrypi:~ $ rs-depth**
> **There are 1 connected RealSense devices.**
> **rs_error was raised when calling rs2_create_device(info_list:0x1cdb6b0, index:0):**
>     **Could not open device. Last Error: Device or resource busy**

ev-mp, 2019-04-14:
@NMovere ,seem the logger fails with the same error, so unfortunately it cannot be utilized.
Please share the kernel log when the error occurs (`dmesg`)

Regarding the previous post : did you try to put a powered USB-hub in-between ?
Also , did you manage to try this camera on other Win/Linux machine, and what were the results?

For me it's the first time I encounter this error message, so I'm not ready yet to seek replacement, at least till we manage to isolate and home on the faulty link.

NMovere, 2019-04-14:
@ev-mp 

After a while, the **rs-fw-logger** started spewing this (not sure if this means it is working or not):

> **Waiting for RealSense device to connect...**
**RealSense error calling rs2_device_hub_wait_for_device(hub:0xf7b368):**
    **hwmon command 0x10 failed. Error type: Invalid Command (-1).**

**Waiting for RealSense device to connect...**
**RealSense device was connected...**
**Device Name: Intel RealSense USB2**
**Device Location:** 

**2019-04-14 10:34:20  FW_Log_Data:0F 00 00 00 A0 02 05 10 50 00 83 F0 CC 24 00 00 00 00 00 00 91 98 3A A7 A0 A1 15 10 90 00 EB D0 01 00 01 00 1E 00 00 00 FA CD EC D5 A0 A1 15 10 90 00 3B E1 01 00 01 00 1E 00 00 00 BB 55 ED D5 A0 A1 15 10 8A 00 EB F0 01 00 01 00 1E 00 00 00 07 AD ED D5 A0 A1 15 10 90 00 EB 00 01 00 01 00 1E 00 00 00 E6 4D B5 EB A0 A1 15 10 90 00 3B 11 01 00 01 00 1E 00 00 00 94 CA B5 EB A0 A1 15 10 8A 00 EB 20 01 00 01 00 1E 00 00 00 A1 1E B6 EB A0 A1 15 10 90 00 EB 30 01 00 01 00 1E 00 00 00 F3 9B B8 EB A0 A1 15 10 90 00 3B 41 01 00 01 00 1E 00 00 00 E2 1F B9 EB A0 A1 15 10 8A 00 EB 50 01 00 01 00 1E 00 00 00...(more numbers)**

NMovere, 2019-04-14:
I have not tried it on a different machine, nor have I used a powered USB hub yet (I will have to purchase one).

**My only other computer is a Macbook laptop with MacOS Mojave Version 10.14** — If not the latest version, close to it. **I am not sure if the SDK is supported on this platform?**

**Here is the long kernel log (dmesg)**:

>pi@raspberrypi:~ $ dmesg
[ 2839.205112] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2839.205328] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2839.206387] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2839.206604] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2839.206616] uvcvideo 1-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2839.206627] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2839.206638] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2839.207075] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2404
[ 2840.208515] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2840.208616] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2840.208828] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2840.209864] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2840.210079] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2840.210092] uvcvideo 1-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2840.210104] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2840.210114] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2840.210560] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2405
[ 2841.222471] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2841.222569] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2841.222801] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2841.223851] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2841.224044] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2841.224057] uvcvideo 1-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2841.224067] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2841.224078] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2841.224505] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2406
[ 2842.225939] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2842.226040] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2842.226255] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2842.227322] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2842.227538] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2842.227551] uvcvideo 1-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2842.227561] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2842.227572] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2842.228014] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2407
[ 2843.230116] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2843.230406] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2843.230617] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2843.232336] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2843.233539] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2843.233569] uvcvideo 1-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2843.233593] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2843.233606] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2843.235999] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2408
[ 2846.237825] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2846.237906] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2846.238119] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2846.239135] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2846.239325] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2846.239339] uvcvideo 1-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 2846.239358] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2846.239368] uvcvideo 1-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 2846.239784] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2409
[ 2847.241183] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[... cut ...]
[ 2968.034181] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input2514
[ 2969.035774] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2969.035870] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 2969.036073] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad6)
[ 2969.037097] uvcvideo: Unable to create debugfs 1-6 directory.
[ 2969.037307] uvcvideo 1-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[ 2969.037319] uvcvideo 1-1.3:1.0: Entity type for

ev-mp, 2019-04-14:
Thanks for the info.
The first section is exactly the type of data required for analysis. 
> 2019-04-14 10:34:20 FW_Log_Data:0F 00 00 00 A0 02 05 10 50 00 83 F0 CC 24 00 00 00 00 00 00 91 98 3A A7 A0 A1 15 10 90 00 EB D0 01 00 01 00 1E 00 00 00 FA CD EC D5 A0 A1 15 10 90 00 3B E1 01 00 01 [...]

Please provide the full log. Redirect to a file if you have a limited terminal size:
>rs-fw-logger > log.txt

NMovere, 2019-04-14:
[RealSense_log.txt](https://github.com/IntelRealSense/librealsense/files/3077750/RealSense_log.txt)

NMovere, 2019-04-14:
@ev-mp 

I was able to get this running on my Macbook (MacOS Mojave 10.14) using the RealSense Viewer. Although, it is giving me a minimal firmware requirement: 

> Current firmware version: 05.09.02.00
Minimal firmware version: 05.10.03.00

The camera in the RealSense Viewer shows only a IR video stream -- there is no depth information. I've tried to update the firmware using my Raspberry Pi (same specs as above; same Raspberry Pi as above), here is what happens (following this Linux tutorial by Intel https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf):

> echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo xenial main' | sudo tee /etc/apt/sources.list.d/realsensepublic.list
> sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
> sudo apt-get update
> sudo apt-get install intel-realsense-dfu* 

**On that last line is where I get this error:**
>Reading package lists... 
>Done Building dependency tree 
>Reading state information... Done
>E: Unable to locate package intel-realsense-dfu*
>E: Couldn't find any package by glob 'intel-realsense-dfu*'
>E: Couldn't find any package by regex 'intel-realsense-dfu*'

ev-mp, 2019-04-15:
@NMovere , the logs indicate that the device is functional, and that the initial issues were probably due to the HW setup.
I ran through the log file and it lacks the log messages produced during the camera's initialization, probably  overwritten by later messages since the FW internal log buffer is limited.

therefore I'd ask you to rerun the logger as follows:
- Disconnect D415.
- Run `rs-fw-logger > log`
- Attach D415.
- Run `rs-enumerate-devices`
- Share the log and the output produced in `rs-enumerate-devices` console.
Sorry for the inconvenience.

Also take a note that the FW version used:
>Current firmware version: 05.09.02.00

is not compatible with Librealsense SDK 2.16.1+ ([Link] (https://github.com/IntelRealSense/librealsense/issues/2611#issuecomment-433679064)) on Linux machines with v4l driver, similar to Raspberry Pi's.
So in any event you'll have to upgrade the FW using one of the methods provided:
-[Windows Guide](https://www.intel.com/content/www/us/en/support/articles/000026587/emerging-technologies/intel-realsense-technology.html)
-[Ubuntu Linux Guide](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf)
Again - sorry but there is no MacOS-compatible tool currently available.
We're actively working to expand the reach and make FW update available on additional OSes, including MacOS. So stay tuned.

NMovere, 2019-04-15:
@ev-mp 

I had it running for about 3 hours and nothing was outputted.

Command ran: "rs-fw-logger > log.txt".

ev-mp, 2019-04-16:
@NMovere hi,
Did you manage to rerun and collect the logs ?

Additionally, while looking at #3770 I saw that the user reporting similar issues with `hwmon error`,  which, I believe related to missing installation of `udev-rules`.
Can you verify `udev-rules` are properly installed on the Raspberry Pi? https://github.com/IntelRealSense/librealsense/issues/3697#issuecomment-482006790

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

@NMovere Any update? Can you verify `udev-rules` are properly installed on the Raspberry Pi? https://github.com/IntelRealSense/librealsense/issues/3697#issuecomment-482006790

And if still have problem, please help collect the logs. Thanks!

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@NMovere How is it going? Did you get through the issue? Looking forward to your update. Thanks!

NMovere, 2019-05-06:
Sorry for the delayed reply, everything is working. Thank you!

closed 2019-05-06 by NMovere