realsenseai/librealsense #3754: Anaglyph 3D RGB images [Feature Request]

MartyG-RealSense, 2019-04-14:
Hi,

When doing some research into the creation of anaglyph 3D images like those viewed with 3D glasses, I wondered if it might be possible to create such a mode in the RealSense SDK by capturing an RGB stream from the left and right IR imagers and aligning them.

Here is a code snippet posted by @dorodnic for getting RGB from the IR imager.

rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, RS2_FORMAT_RGB8);
pipe.start(cfg);

Presumably, the left and right image could be got by using index numbers for the streams, like with the code example for getting left and right IR:

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers

I imagine the anaglyph result would be something like the short YouTube video below once the color channels had been adjusted.  It could be an alternative to the depth image that is created from the left and right IR images.

https://www.youtube.com/watch?v=9UawGZLPrIU

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

Hi @MartyG-RealSense,
Thank you for your request.
Please allow me to briefly explain that not all D400 series camera can output RGB from the IR imager.

> capturing an RGB stream from the left and right IR imagers and aligning them

For the RealSense D41x,  the RGB stream is available only via the "left" IR imager.
Left and right IR imager on the D43x series camera are monochrome sensors; hence, no RGB color.

MartyG-RealSense, 2019-04-28:
Thank you very much for the technical clarification!

closed 2019-04-28 by MartyG-RealSense

awatson3, 2019-09-26:
@MartyG-RealSense Thank you

@RealSenseCustomerSupport Per above comments, there are no Realsense models that support simultaneous RGB streams from Left and Right cameras, correct?

MartyG-RealSense, 2019-09-26:
@awatson3  Your question above is more likely to be seen if you post it on your own message, as this case here is now closed and the Intel staff won't be tracking it.  Thanks!