realsenseai/librealsense #3756: T265 Ubuntu Mate 18.04.2 Realsense Viewer camera stream error

sirisian, 2019-04-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5448 | 
| Operating System & Version |   Ubuntu Mate 18.04.2  | 
| Kernel Version (Linux Only)    |  4.15.0-1033-raspi2                                         | 
| Platform                                 | Raspberry Pi 3B+ |
| SDK Version                            |  2.20.0                         | 
| Language                            |  None                          | 
| Segment			|                            | 

### Issue Description

I built the SDK on the latest 2.20.0 tag in release.

I should mention that I can just turn off both the fisheye cameras in the Realsense Viewer and I can still view the pose information, so this seems just related to pulling back the streams. (Everything else seems to work fine, so it's relatively minor).

When I click turn on with fisheye cameras selected:

15:07:10.463 [23785] [E] Device-8F50: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 240128
15:07:10.668 [23777] [E] Device-8F50: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
15:07:10.668 [23777] [E] Device-8F50: Entered state [ERROR_STATE]

The toggle turns off in the viewer.

Filmengineers, 2019-04-17:
That's probably because of the USB bandwidth bottleneck. It requires a USB 3.0 to stream camera feeds I guess.

P.S. Would you mind sharing the pi installation instructions for librealsense on Ubuntu?

Thanks,

sirisian, 2019-04-18:
@Filmengineers I've included the node.js steps also:

Go to https://ubuntu-mate.org/download/ and download Raspberry Pi Experimental 18.04.2
Download etcher and write the image to an SD card.
Put it into the Raspberry Pi and turn it on.

Depending on monitor the resolution will be correct or it'll be wrong. If it's too small and the welcome screen is small you'll need to press tab to navigate to the hidden next buttons. After you finish the welcome screen you can press ctrl+alt+t to open the terminal.
If you see black bars around the screen then run:
```sudo raspi-config;```
Go to Advanced Options -> Overscan and disable

If your resolution is broken and you don't see the option in the raspi-config to change it to 1080p then edit:
```/boot/config.txt```
And set:
```hdmi_group=1```
And:
```hdmi_mode=5```
The 5 can be whatever resolution you want. The Pi doesn't support random ones, not sure why. It'll fallback to something that works usually.

There's a bug in the Ubuntu Mate beta image where openssh-server is setup wrong. If ssh doesn't work and you need it run:
```sudo dpkg-reconfigure openssh-server;```

To build librealsense run:
```
sudo apt-get update;
sudo apt-get upgrade;
sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev npm nodejs;
cd ~/Desktop;
git clone https://github.com/IntelRealSense/librealsense.git;
cd librealsense;
git checkout tags/v2.20.0;
./scripts/setup_udev_rules.sh;
mkdir build;
cd build;
cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_NODEJS_BINDINGS=1;
sudo make uninstall && make clean && make && sudo make install;
cd ../wrappers/nodejs;
npm install;
cd examples;
npm install;
```

Done. Plug in your T265 sensor and run:
```rs-pose;```

In theory you can also run the node.js program with:
```
cd ~/Desktop/librealsense/wrappers/nodejs/examples;
node nodejs-capture.js;
```

It apparently does nothing with T265 though. It just segmentation faults when creating the context for me.

mvanderbeek123, 2019-05-12:
@sirisian were you able to resolve your issue? I am using the t265 camera on an odroid with USB3.0 and running the program through ROS with the rs_t265.launch file. I have the same error message and cant seem to find a good solution. Any clue is welcome !

sirisian, 2019-05-13:
@mvanderbeek123 In my case the other poster is probably right and it's a USB 2.0 bandwidth issue for me. If you're getting the error with USB 3.0 then that seems to indicate this is not related to bandwidth. I took a break from playing with the device. This is probably something only the Intel devs can solve.

closed 2019-05-13 by RealSenseCustomerSupport