realsenseai/librealsense #3758: T265 UWP

Carlogeno, 2019-04-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   |               | 
| Operating System & Version |   Win (10)   | 
| Platform                                 | PC  |
| SDK Version                            | 2.2                        | 
| Language                            |  unity                           | 
| Segment			|                    | 

### Issue Description
Building UWP project from unity plugin throw exception form RealSense.NativeMethods

MarshalDirectiveException Cannot marshal type "System.Object"

Carlogeno, 2019-04-16:
Anyone know if T265 should work on UWP builds? I can't find information about it.
Thanks in advance

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

@Carlogeno
There is not a UWP (DMFT) release for T265

closed 2019-05-13 by RealSenseCustomerSupport